opentx/opentx #8350: Timer display collides with model picture when exceeding 99:99

AchimSto, 2021-03-09:
The display of a flight timer and a total flight timer together with the model picture works very fine until the total flight timer exceeds 99:99. Then it continues with 100:00 and the last digit is covered partially by the model picture.
I would expect that (every?) timer would change its (configurable) behaviour in this situation from mmm:ss to hh:mm.
Is this already implemented? I couldn't find any settings for this.

[image]

**General information**
 - OpenTX version 2.3.10-otx
 - Hardware FrSky X10s express

3djc, 2021-03-09:
Use timer widget ?

AchimSto, 2021-03-09:
> Use timer widget ?

Yes. I use the timer widgets and unfortunately I cannot change any settings according format or so. Do I have to create a widget by myself?

3djc, 2021-03-09:
Thats not the timer widget

AchimSto, 2021-03-09:
> Thats not the timer widget

Oh! When I press the TELE button I enter the Main view 1 layout site. Here I can select the layout and several widgets. In "setup widgets" Ican choose a timer widget und change its source an colour. Here it would be great if I could choose the time format when time exceeds 99:99.

closed 2022-04-17 by stale[bot]