opentx/opentx #8355: Companion Diff Feature

MartyHugh, 2021-03-10:
What would be nice would be to see what changes you have made since your last good backup
a) in case any unusual settings (e.g. bad trim setting) or indeed any setting which has been accidentally changed.
b) so that where a backup/cloned radio is being maintained, so that any valid changes can be checked before updating and synchronising

A nice solution would be the ability for Companian to be able to select any two .otx settings files (or in memory loaded settings files) and produce a diff summary of the differences between them.

Further functionality would allow changes to selected one at a time to copy one way or the other.

elecpower, 2021-03-10:
@MartyHugh Companion has a Compare Models feature that colour highlights differences though not as compact as a diff.

MartyHugh, 2021-03-10:
@elecpower thank you for replying, that feature looks useful.  

However. What would be nice would be to compare complete files though.

closed 2022-04-17 by stale[bot]