opentx/opentx #8360: SI and SJ port

olar33, 2021-03-12:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

Hello,
I installed RC Technik handles with 2-positions sticks plugged into the SI and SJ ports. My radio is a Horus X10S express with open TX 2.3.11
The switches do not work.
My question are the SI and SJ ports taken into account under openTX?
Thanks for your help.
Olivier.

kilrah, 2021-03-12:
Need to enable them in hardware menu.

olar33, 2021-03-20:
Thanks for your amswer. I have activated the 2 ports in the hardware menu. After a test all the switches work except the SI and SJ ports