opentx/opentx #8362: Trainer input on heartbeat pin Horus after external ACCESS mod

EmanuelStassar, 2021-03-13:
MPM has a new feature. It now supports sending trainer data over the heartbeat pin of the module connection. On Taranis this works, but on Horus the heartbeat pin is not connected by default. The external ACCESS mod uses the heartbeat pin. 

Would it be possible to use this this hardware mod to enable trainer input as well? It would be really helpful as trainer input over S.port does not work on Frsky Transmitters.

MRC3742, 2021-03-14:
You would first need to do the hardware mod inside the external Multi Module.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/CPPM_HW_Mod.md

Next their is a post with pictures on RCG that has done a mod inside a Horus X12 to do just what your looking for?
I believe it is using the trainer port option but wired inside the TX bypassing the need to plug in to that port.
https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module&p=46695615#post46695615

Rich

EmanuelStassar, 2021-03-14:
Hi Rich, 
You are correct, a small mod inside the MPM is needed. 
I posted the info about modding the internals of the X12. I later found out my mod is not possible if you als want the external access mod. That is why I opened this request.

MRC3742, 2021-03-14:
I didn't realize that was your post as you never reported that you were having problems using that mod together with the external access mod? (For R9M ACCESS?)
I suppose that mod on RCG would work if you were not interested in using the external access mod? 

I'm not sure what your asking to do with this issue then? All 5 pins already seem to be in use.

Rich

EmanuelStassar, 2021-03-14:
OpenTx support trainer input on the heartbeat pin in some transmitters, but not Horus. My request is to enable this option in the trainer settings if the Horus has the external ACCESS mod which connects the heartbeat pin to a serial ports if I understand the mod correctly. I hope this is possible from a hardware and software point of view.

MRC3742, 2021-03-15:
That mod looks like the heartbeat pin is connected to the serial port using an additional inverter on the Horus Radios.
Not sure if that's possible but maybe one of the devs can answer that variable.
https://github.com/opentx/opentx/wiki/Externalaccessmod

I also have an open issue #8370 to enable it on the Taranis non Horus models.

I hope both issues are able to succeed and be merged into the code - Rich

brainbubblersbest, 2021-05-09:
Look at #5518 at the end 😉
The PR is open a few Days now. But your idea Was already discussed. 🙂

closed 2022-04-16 by stale[bot]