opentx/opentx #8369: Cancel/Undo value change

drtrigon, 2021-03-14:
Currently when I use otx on TX16S and a edity a value the new value is stored no matter whether I press "ENT" or "RTN" afterwards. I would suppose to store the new value on "ENT" only and restore the value before editing if I press "RTN".

**Describe the solution you'd like**
I select a field having value A and press the roller button (on TX16S) in order to edit the value. I use the roller button to choose value B still beeing in edit mode. Currently it does not matter if I press "ENT" or "RTN" now as in any case the new value B gets set. I would like to see if I finish the edit mode by pressing "RTN" the edit should be "canceled" and the old value before editing A should be restored.

**Describe alternatives you've considered**
-

**Additional context**
-

kilrah, 2021-03-14:
Not going to change that since that's how every R/C radio out there works.

drtrigon, 2021-03-14:
Yeah, I assumed that and thus the answer was expected.

However may be it is time to change the fact that every radio out there works like that - I expect otx to be better than the others... ;)

kilrah, 2021-03-14:
I don't see how it could be better, it would cause massive confusion with users especially since everything IT is moving to a similar "instant save" model nowadays.

pfeerick, 2021-03-15:
Not so... if you are changing an option, in most cases, in IT, it does not save until you press the OK/Apply/Save button. 

I don't see where the confusion would lie... back = undo / don't apply, enter = apply. And saying it's like every other radio out there is like saying "I dunno why we bothered making OpenTX, since it's nothing like every other radio out there!" :laughing: 

On my old Spektrum DX6i, I could understand not being able to undo... hard to have a back button when your only menu input was a roller + button. Not the case with most OTX transmitters. That is one particular thing I like about the OrangeRx transmitter software... you have to press and hold to save any parameter changes... if you change your mind, just get out of the menu, nothing is changed. Would be much better executed if it were using the back and enter keys properly, but at least it was thought of.

kilrah, 2021-03-15:
>Not so... if you are changing an option, in most cases, in IT, it does not save until you press the OK/Apply/Save button.

On old style stuff yes, but on all "modern" apps it's instant, see google docs for example, there isn't even a "save" feature.

>  And saying it's like every other radio out there is like saying "I dunno why we bothered making OpenTX, since it's nothing like every other radio out there!" 😆

It would be changing something significant OpenTX users have been used to for 8 years...

drtrigon, 2021-03-15:
> On old style stuff yes, but on all "modern" apps it's instant, see google docs for example, there isn't even a "save" feature.

Not everything google does is perfect or better than others.

Nowadays a lot of software is going to work/look/feel like smart phone apps. Personally I consider this a step backwards and not an improvement because it tends to simplify things up to a degree where it becomes totaly useless. We are talking about a security relevant piece of software that controls a real thing beeing mid in air and able to fall down to someones head. We are not talking about a shiny bling-bling web app or game.

I also do not like arguments like "it is like that because it was like that all the time". I used to work with Futaba FC-28 e.g. but the argument otx should be exactly like that firmware as it was like that in the old days sounds wrong to me.

I was several time in the situation where I was just trying around in otx and in the end wanted to restore the value as it was, but as I was also thinking about the model, implications and considered a lot of other stuff I was just not sure what setting I had before. In that situation - remember its security relevant - I would have been happy to have this "undo/cancel" behaviour. Of course I could have gone to my PC, open companion and check there - but honestly that is just inconvenient and feels very pedestrian-like.

kilrah, 2021-03-15:
>  We are talking about a security relevant piece of software that controls a real thing beeing mid in air and able to fall down to someones head. We are not talking about a shiny bling-bling web app or game.

That's the whole point, it's about safety, and the last thing we'd want to hear is someone who crashed because they changed something, pressed RTN as usual and the value reverted on them unexpectedly/becasue they forgot/dodn't know it worked that way.

> I was several time in the situation where I was just trying around in otx and in the end wanted to restore the value as it was, but as I was also thinking about the model, implications and considered a lot of other stuff I was just not sure what setting I had before.

Write it down, make a backup/copy of the model before changing such stuff, whatever... but this really seems like a case where you need to find the proper way to work with things how they are instead of expecting the software, and >100K users to adapt to you.

jasonleekennedy, 2021-03-15:
who makes settings changes when their craft is in the air?  shouldn't SOP be to make the changes on the ground, test everything works as expected (on the ground) then fly?

drtrigon, 2021-03-15:
RTN is not a button you press by accident but intentionally if you want to undo or cancel an edit.

I think we miss the point here; it's not about forcing people to something but about moving forward and adopt to new hardware (that has a RTN button e.g.).

I am an inclusionist and do not see the taboo or danger related to this proposal.

However may be you want to consider adding a global option that allows to enable the proposed behaviour.

pfeerick, 2021-03-15:
> On old style stuff yes, but on all "modern" apps it's instant, see google docs for example, there isn't even a "save" feature.

I'm not quite sure how a 'responsive' website design relates to radio transmitters... but ok. But there is both an OK and Cancel button on the page setup Google Docs page... why do they need the cancel button if people don't make mistakes , change their mind, or preview things, and then choose not to save them if they don't like the preview? Lets not beat that dead horse with out of context comparisons. ;) 

> However may be you want to consider adding a global option that allows to enable the proposed behaviour.

Took the words right out of my mouth... if there truly is an issue, rather than a perceived issue, and this improvement to usability implemented, have it behind a global option. Options like writing it down, or backing up models just because you might be changing a setting to see what it looks like, and then deciding it doesn't work and wanting to undo it are completely missing the point that technology is supposed to be improving things, not going backwards.

lshems, 2021-03-16:
I would like pagewide roll back option. Not because changes may not be instant, but because the GUI is to easy to do things you don't want and to difficult to see WHAT it did you didn't want.

Select a line and accidentally press up or down->Confusion->enter->some unwanted change you cannot easily detect.

Proposal: highlight all changes, and ask confirm on exit page.

Yes, you should always make a backup first, before changing anything. 
Counterproposal: MAKE a backup after exiting page that has changed, automatically. Keep the last 10 to restore from. Since it stores after the MOMENT OF CHANGE, it is easy to find the version before you started to fiddle around. It will probably be of yesterday.

lshems, 2021-03-16:
> > who makes settings changes when their craft is in the air?
> 
> The willing risk taker who _wants_ to see their craft crash. 😆 We're not talking about altering trim settings, after all!

Sorry, but complex glider setups ask for much more than just a setting to change DURING flight. Most of it preferably by trims or other switch combinations.
But setting an elevator compensation curve is done by flying and applying different flaps and adjust the points on the curve. I have some nice LUA script to do so safely, but it IS common practice.
An UNDO is not the right thing there. It is IN FLIGHT adjustment that needs to be active directly.

So I feel strongly about my backup rollback list of automatic rollbacks in a special folder outside of the main view.
Scroll through the available backups on the model selection page in a free slot to restore. :)

pfeerick, 2021-03-17:
> complex glider setups

Ah, see, you damned your argument right there... *complex* :stuck_out_tongue: 

There's never any 100% perfect solution... just as that lovely saying... what is it... *exception to the rule* ... says. :wink: 

But, thinking about this more, and also playing with the Radiomaster TX16S more (since I've upgraded from 9X/9XR PRO OpenTx due to them no longer being actively supported), I can see one potential problem with this. Enter is under the roller, making the Return button easier to press... thus the habit is formed to press Return after changing a value, rather than pressing the roller for Enter. In other words, an 'return = restore, enter = save' behaviour may be more suited to transmitters with dedicated 'return/enter' buttons, maybe not even present on six button transmitters (unless incorporated into the 'menu' return button behaviour), and awkward with roller interface enter transmitters. Potentially making things inconsistent - never a good start.

I think this is a legacy issue caused by moving from transmitters like the 9XR/9XR Pro to the Taranis, where there are separate exit and enter buttons (thus opening the option for different behaviours rather than the same)... it should have been done then... but is too late now, unless it is toggled by a global option. To do otherwise will create the same issues/division Mode 1/Mode 2 has for mostly the same reason... incremental change/improvement resulting in poor overall design choices... it happens all the time, and probably always will.  I'm not entering into the whole "Mode 2 is better than Mode 1 / Mode 1 is better than Mode 2" argument... each has their pros and cons... as do Modes 3 & 4... I'm just using it as an comparison of a decision/change not made causing problems later which most hobbyists who know the history of transmitters should be familiar with. 

Learning from that I would suggest it leaves two options - leave it as it is (either permanently, or until the next major version as a major behavioural change) or implement it it with a configuration flag to enable behaviour if wanted (which could be toggled on by default at a future date/release, allowing old behaviour to to be restored if desired). I know the developers would favour the prior option... hey, it's easier! :smile: ... but I'd really like to see the latter. Perhaps if there is enough interest, or someone with enough knowledge of the OpenTx source puts together a PR implementing this .... ?

Having said that, I'll shut up now... (and look into this auto-backup thing, as it sounds interesting) ... this has gotten longer than what I intended!

mpjf01, 2021-03-29:
I'm thinking of a couple of examples that I use commonly. 

Editing a name. At present, one cycles through the characters until the one needed is visible and hits ENTER. That character is saved and the cursor moves to the next spot where the next character is selected...and so on. Are you saying you want the previous name saved somewhere and that the new name only becomes active when another key is pressed? If so, why? Surely the user has enough opportunities to decide that what he is doing is wrong while he is doing it. 

Then I'm thinking about changing values in a curve. At present I change the value of each point in a curve while watching the movement of the relevant control surface. So I want the new value active immediately. When I have the value I want I hit enter which saves the value. If I have to hit enter to save it before seeing the effect on the model it would mean many, maybe hundreds, of extra hits on the enter key to go in and out of edit mode, change the value, repeat, repeat until it is right. 

If, on the other hand, your suggestion is to have an extra step in the process so that when switching away from the curve to another screen (same would apply with every screen presumably) you have to confirm that the curve needs to be saved because you have altered a field in it then that would seem to require the system to keep the image of the screen before editing so that it could be reverted to should the user decide to abandon what he has just done. That sounds like it might be a fair bit of work for the developers. 

In these two examples I don't see any real benefits. Have I misunderstood the request, and if not, what is an example where we would be better off?

pfeerick, 2021-03-30:
My interest was only in a single value change undo, but I can see the benefits of a whole page undo when making multiple changes. But I think that is more within the realms of using backup options, rather than adding that sort of complexity to OpenTx. 

> Editing a name. At present, one cycles through the characters until the one needed is visible and hits ENTER. That character is saved and the cursor moves to the next spot where the next character [...]

In my mind, I would see it as if you press enter, the character is kept, if you press return, the character is lost. So there is no need to save the entire string... unless the behaviour was multi-input was only saved when you press and hold enter, and lost on a return. I'm not sure the best answer to that one... the first is would be the more 'consistent', the second more 'logical'. 

> Then I'm thinking about changing values in a curve. At present I change the value of each point in a curve while watching the movement of the relevant control surface. So I want the new value active [...]

This is where it gets tricky... as a curve by definition is also multi-input... hence, allowing easy reversion of the entire curve (input) to a single 'field' makes sense. For editing something like a curve, maybe, just with the naming multi-input.. press and hold to save, but the changes are applied/visible until you do that... but for a multi-input... this would probably also require return a  prompt of 'Do you want to keep unsaved changes?' ... as that momentary brain fart moment when you hit return instead of enter will annoy you no end. 

Yes, it would be a bit of work on the developers, but at the end of the day, the whole project is. That's why this discussion is good, to find out the merit of such a feature, what it would look like, if it has enough value to be worth the effort, or is impractical.

mpjf01, 2021-03-31:
> In my mind, I would see it as if you press enter, the character is kept, if you press return, the character is lost. So there is no need to save the entire string... unless the behaviour was [...]

At the moment the main annoyance is if you select the wrong character in a string you have to cycle back through the string to change it. There's no way to go backwards once you press enter. Most often it's pressing the enter that creates the problem if using the scroll wheel or roller as this also acts as the enter "button" and when pressing it it is too easy to rotate it a little and change the character. Your first option doesn't change that, and the second would still require cycling through the characters to fix the faulty one before saving the string. I'm not sure how users would be better off than they are now. 

I would prefer it if you could use the roller to cycle through the string, either way back or forwards, press enter to select the character to be changed or the "slot" to add one, select the correct value as now, press enter to save that and get out of edit mode for that character.

> This is where it gets tricky... as a curve by definition is also multi-input... hence, allowing easy reversion of the entire curve (input) to a single 'field' makes sense. For editing something like [...]

Using a multi point curve to match, say, travel between left and right flap, requires a lot of selection and value changing of each of the many curve points in an iterative process. Once you have got that right and the data has automatically been saved I can't imagine a situation where you would want to revert to a previous state (ie a saved state prior to the edit). It would be quite rare I suggest. But a prompt before leaving the screen asking the user to save the changes (Yes/No) would not be hard for users to deal with and being able to revert might have occasional benefits.

lshems, 2021-04-01:
Face it. The opentx GUI is clumsy and error prone. That's by design. Selecting a line and immediately create a copy on up or down is very fast, but clumsy.

It's a trade-off between fast and clumsy, and slower but more intuitive.

lshems, 2021-04-01:
Not criticizing, just observations.

closed 2022-04-16 by stale[bot]