opentx/opentx #8370: TRAINER for Taranis Radios (except Horus) with External MULTI selected - allow Master/CPPM to be available

MRC3742, 2021-03-14:
With Multi Module version 1.3.2.61 and above, a new feature has been added to allow the external module, using an RX protocol, to output CPPM to the heartbeat pin. This requires a MPM internal resistor mod for connection to USART.TX1 pin.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/CPPM_HW_Mod.md
This option is now working on Taranis radios (not Horus) using internal XJT Modules and running erskyTX.

Currently in OpenTX if External MULTI is selected on these radios the only options for trainer are Master/Jack and Slave/Jack
When the trainer Master/CPPM is selected, the only option for External RF is "OFF"

Is it possible to enable the Master/CPPM option when the External MULTI is selected?
This would allow wireless trainer via MPM to a Taranis Radio using internal XJT for model control.

Thanks - Rich

MRC3742, 2021-09-12:
I am commenting to prevent closure.
Is there is any interest to allow this function for Taranis Radios?

akschu, 2023-05-29:
This is affecting me too