opentx/opentx #8371: MODEL MEMORY SWITCHING

WOODY1958UK, 2021-03-15:
HI -  FIRST PLEASE EXCUSE CAPS - BUT I AM PARTIALLY SIGHTED - SO I HAVE A BIG IDEA TO IMPROVE OPEN TX - AND I DON'T KNOW WHY IT DOESN'T DO IT - SO I WANT TO BE ABLE TO SWITCH MODEL MEMORIES IN FLIGHT - I CAN DO THIS WITH MY OLD MULTIPLEX 3030 - AND IT'S BRILLIANT - SO AT THE FLICK OF A SWITCH I CAN CHANGE MIXES / RATES / OUTPUTS / STICK INPUTS / TRIMS - EVERYTHING - SO IDEA - WHEN I USED TO FLY F5F WHICH IS ELECTRIC THERMAL FLYING - I WOULD HAVE LAUNCH MEMORY WHERE THROTTLE IS ON LEFT HAND STICK - WITH RATES SET AS NEESSARY - FLAPS MAYBE ON A THREE POSITION SWITCH - THEN AFTER LAUNCH PHASE I COULD SWITCH MODEL MEMORY AND FLAPS WENT TO THE THROTTLE STICK WITH SPEED UP FLAP AT TOP AND A THERMAL SETTING AT BOTTOM WITH A BLANK SPOT IN THE MIDDLE - THEN SWITCH TO LANDING PHASE WHICH SWITCHES OUT ANY MIXES APART FROM CROW - FLAPS / AILERON / ELEVATOR MIX - DOES THAT ALL MAKE SENSE - NEEDS A BIT MORE THINKING ABOUT - BUT IF YOU WANT TO SET UP ANOTHER MODEL YOU CAN EASILY COPY THE LAUCH - CRUISE - LANDING PHASES - WOULD BE MUCH EASIER THAN THE MIXING I HAVE SET UP ON MY FRSKY X9D+ SE 2019 WHICH TOOK ME TWO DAYS TO PROGRAM WHAT I WANTED AND I AM STILL NOT 100 % HAPPY WITH IT - WHAT DO YOU THINK - COULD SOMEBODY PLEASE GET BACK TO ME - WOULD BE GREAT TO TALK TO SOMEBODY - MUCH THANKS - DAVID

kilrah, 2021-03-15:
In OpenTX you can do all of this without changing model memory using flight modes.

closed 2021-03-15 by kilrah

WOODY1958UK, 2021-03-15:
HI ANDRE I HAVE TRIED THIS AND CAN'T GET IT TO WORK I HAVE MANAGED TO DO WHAT I WANT WITH SOME VERY COMPLICATED PROGRAMMING IN THE INPUTS / MIXING PAGES BUT IT'S NOT 100 % - AND I AM STRUGGLING TO CHANGE THE INPUTS TO DO EXACTLY WHAT I WANT THE FLIGHT MODES WILL ALLOW ME TO HAVE DIFFERENT TRIM SETTINGS - BUT FLIGHT MODES WON'T ALLOW ME TO CHANGE STICK ALLOCATION OR RATES OR REFRESH THE MIXING PAGE IT'S NOT AS SIMPLE AS THAT - AS AN EX COMPETION ELECTRIC GLIDER FLYER I COULD DO ALL THIS ON MY VERY OLD MULTIPLEX 3030 WHICH ONCE I HAD REALISED WHAT I COULD DO - I WENT WOW I HAVE WATCHED MANY YOU TUBE VIDEOS AND READ THE TRANSMITTER MANUAL - ALL OF WHICH DON'T HELP COULD YOU POINT ME IN THE DIRECTION 0F WHERE IN OPEN TX I CAN FIND WHAT I AM LOOKING FOR THANKS FOR YOUR HELP GETTING BACK TO ME IS SO APPRECIATED REGARDS DAVID      DAVID BRIAN HOMEWOOD FLAT 32, MARGARET HOUSE LEALANDS DRIVE UCKFIELD EAST SUSSEX ENGLAND TN22 1FH

lshems, 2021-03-15:
DAVID?

SINCE YOU ARE CLEARLY STRUGGLING I PROPOSE YOU A FREE LICENSE TO ALL THE JUSTFLY APPS AND MODEL SETUPS. I THINK THEY DO EXACTLY WHAT YOU WANT WITHOUT COMPLEX PROGRAMMING. JUST CONTACT ME BY SENDING A MAIL TO JUSTFLYSOLUTIONS AT THE GENERAL GMAIL.COM

jasonleekennedy, 2021-03-15:
inputs can be conditional, so you can setup multiple inputs, then have the inputs active based on a condition based on your flight mode, which will let you change around what stick effect what behavior farther down

https://doc.open-tx.org/manual-for-opentx-2-2/software-overview/model_menus/inputs

looks like mixers can also be conditional
https://doc.open-tx.org/manual-for-opentx-2-2/software-overview/model_menus/mixer

between those 2 items, you should be able to setup just about any 'modes' you want

WOODY1958UK, 2021-03-15:
HI GUYS THANKS FOR GETTING BACK TO ME SO QUICKLY - I HOPE I HAVE MADE SENSE WITH WHAT I WANTED TO DO SO PLAY TIME - HE HE HE - HOPEFULLY I CAN GET TO WHERE I WANT TO - I WILL KEEP YOU POSTED ONE ISSUEN THAT IS MY FAULT IS THAT I BRING TO OPEN TX PREVIOUS TRANSMITTERS THAT COULD DO EXACTLY WHAT I WANTED AND AM TRYING TO GET THE TRANSMITTER TO DO THE SAME THING IN THE SAME WAY SO THANKYOU FOR POINTING ME IN THE DIRECTION I NEEDED TO GO MUCH APPRECIATED THANKS DAVID      DAVID BRIAN HOMEWOOD FLAT 32, MARGARET HOUSE LEALANDS DRIVE UCKFIELD EAST SUSSEX ENGLAND TN22 1FH  

https://doc.open-tx.org/manual-for-opentx-2-2/software-overview/model_menus/inputs 

https://doc.open-tx.org/manual-for-opentx-2-2/software-overview/model_menus/mixer