opentx/opentx #8373: LUA - model.setFlightMode() problem with trimValue

loetefix, 2021-03-15:
Hi,
There seems to be a problem with model.setFlightMode().

It is not possible to set a negative trimValue with the LUA command model.setFlightMode().
positive Values work fine.

lshems, 2021-03-15:
Never noticed you can change trim values. Nice! When was this added?

Pity it has a bug!

lshems, 2021-03-15:
Nice confusion by the way: getFlightMode() and model.getFlightMode(). Might cause some headaches. But thatnks for adding the model.getflightmode and model.setflightmode. Will be useful if it works.
Finally allows to build a proper HELI or Glider Wizard using flightmodes. :)

lshems, 2021-03-15:
model.getFlightMode(getFlightMode()) to get the current flightmode (sorry, model flightmode) :)

lshems, 2021-03-15:
OK, now some real comments in relation to the 'bug'. For me it seems to work on Taranis XD09+ in sim on OpenTX 2.3.11 ??

loetefix, 2021-03-15:
I testet with tx16s in real and in sim with 2.3.11.
When i set values smaller 0 the trim was set to +100

lshems, 2021-03-15:
Yep, sorry, indeed confirmed. and not to 100, but 125 (yes, max, but for trims thats not 100).

Also, it seems to work only for rudder (Index 0). Not for the other indices??

So we can set trimValues O from 0-125, all the rest doesn't work.

lshems, 2021-03-15:
trying to write -1 at trimvalues[0] results in  a bigger value that then is maxed on 125. Writing 126 or bigger is perfectly possible and also returns that value with model.GetFlightMode.

lshems, 2021-03-15:
Not so nice. Can crash models :(.

lshems, 2021-03-15:
By the way, is the trimsValues order dependant on the channel order, or ...

I see: 
[0] rudder
[1] elevator
[2] throttle

jasonleekennedy, 2021-03-15:
what value ends up if you write -1? sounds like it might be using a signed/non-signed number conversion issue.

lshems, 2021-03-15:
100 or 125 will depend on extended trims probably

lshems, 2021-03-15:
I tried, but it flips really quickly to the max of 125 again.

lshems, 2021-03-15:
Looks like something just above 1000??

lshems, 2021-03-15:
perhaps I can "read" the code. GET seems to be working, so really surprîsed about SET

lshems, 2021-03-15:
going to check github. :)

jasonleekennedy, 2021-03-15:
for a 10bit int -1 signed is the same bit string as a unsigned 1023 (IIRC... been a while)... so it sounds like that might be something going bad between signed/unsigned values

lshems, 2021-03-15:
Well, the GET just reads the values and pushes them in a table to LUA.
The SET does an offset of 1024 (0x3FF). ????

C remaisn abacadabra for me. :(

jasonleekennedy, 2021-03-15:
no that isn't an offset, that is mask which would removes the leading bits.

now this line: https://github.com/opentx/opentx/blob/2.3/radio/src/lua/api_model.cpp#L476

looking at some lua api documentation http://www.lua.org/manual/5.1/manual.html makes me wonder if that luaL_checkinteger should be luaL_checkint (so we get an int out vs a lua_Integer).  But I don't know the lua / c++ bindings to know if my guess is right.

jasonleekennedy, 2021-03-16:
ok, nope, looks like luaL_checkint is just a macro for (int)luaL_checkInteger.

but  think this is it:

https://github.com/opentx/opentx/blob/2.3/radio/src/lua/api_model.cpp#L478

negative numbers in binary are going to be `1xxx xxxx xxxx xxxx`, but when you mask it by 0x3FF (0000 0011 1111 1111), you whip off the sign bit, making -1 [0xFFFF] end up being the max number [0x3FF].

That line needs to accommodate negative numbers

jasonleekennedy, 2021-03-16:
if we just want to clamp to the range [-1023, 1023], something like this works, but this does change the functional behavior for positive numbers also (as masking it would cause 1024 to be come 0 instead of 1023).  Haven't read the rest of the code base to ID if there is a more in code base method of doing this.

change
```cpp
        if (idx < NUM_TRIMS)
          fm->trim[idx].value = (val & 0x3FF);
```

to

```cpp
        if (idx < NUM_TRIMS) {
          if(val > 0x3FF) val = 0x3FF;  //1023
          if(val < 0xFFFFFC01) val = 0xFFFFFC01; //-1023
          fm->trim[idx].value = val;
        }
```

lshems, 2021-03-16:
But that doesn't address the issue that it only seems to work for trim 0?

loetefix, 2021-03-16:
it doesn´t work only work for trim 0.
if you read :

local fmodes = model.getFlightMode(1)

modify by:

fmodes.trimsValues={120,0,120,0,120,10}

and write back by:

model.setFlightMode(1,fmodes)

in my case it works.

lshems, 2021-03-16:
But that isnit according to spec. You should be able to write just one trim value, without reading it first. And that doesn't work.

loetefix, 2021-03-16:
ok. that´s right.but reading it first is not a big problem for me If you want to write only one value.....

local fmodes = model.getFlightMode(1)
fmodes.trimsValues[3]=56
model.setFlightMode(1,fmodes)

thats ok for me.

lshems, 2021-03-16:
I don't care either.its just not according to spec.

I have my own functions where I combine set and get.

Sort of:

Local parameter=function('parameter', newValue)
  Local value =Model.getparameter()
  Value = value or newValue
  model.setparameter(value)
  Return Model.getparameter()
End

This ensures I know what is the updated value, even if it didn't update as I expected. Limit checks I don't know of, ....
  
   Return getParameter

jasonleekennedy, 2021-03-16:
> But that doesn't address the issue that it only seems to work for trim 0?

the loop around it is likely why you have to set all of the trims
https://github.com/opentx/opentx/blob/2.3/radio/src/lua/api_model.cpp#L475

```cpp
      uint8_t idx = 0;
      for (lua_pushnil(L); lua_next(L, -2); lua_pop(L, 1), idx++) {
```
each value that is popped is increasing the idx by 1, the loop isn't using any index value passed by the lua code.   If the lua code passes indexes for sparsely populated arrays, then reading that index here instead of incrementing an index would fix the issue.  if it doesn't, I suspect you would have to read, change, set to get it to work.

lshems, 2021-03-16:
They should uus a loop to check all possible entries.

For I=1,4

lshems, 2021-04-01:
For I=1,6 for 6 trim radios!!

lshems, 2021-04-29:
Not working in 2.4.12 nightly yet??

This is a bug, not a feature request, and can cause big trouble when not fixed.

Your call. not pushing, just posing facts.

closed 2022-04-16 by stale[bot]

lshems, 2022-04-16:
Bravo