opentx/opentx #8376: Add switch options to EX1 and EX2 (Horus X10)

Hobby4life, 2021-03-16:
I Like to request switch options to EX1 & EX2 on the Horus X10

The only possible selection for EX1 & EX2 now are:

- POT
- POT with dent
- MULTIPOS

The inputs can be used for switches too.

Adding the following options would be a great added value to these inputs

- TOGGLE (external pullup resistor needed)
- 2 POS
- MULTIPOS (user defined positions)

https://hobby4life.nl/?p=504

kilrah, 2021-03-16:
Multipos includes 2-pos.

Squirrel05, 2021-03-16:
Yes, so we can do a multiple pos switch.

Hobby4life, 2021-03-16:
> Multipos includes 2-pos.

I dont think you understand what I mean.

Like a 2 position switch **EX1↑,EX1↓,EX2↑ and EX2↓**

EX1 and EX2 now are not selectable as this as behaviour.

kilrah, 2021-03-16:
When you set to multipos and calibrate your 2 positions you'll have EX11, EX12, !EX11, !EX12. Same thing.

Hobby4life, 2021-03-16:
I have tried that.. but it wont give me these switches.

I got EX1 and **EX1**
same for EX2

Also they dont act like switches.
Something is not working properly.

closed 2022-04-17 by stale[bot]