opentx/opentx #8377: Gets stuck in MULTI Assan under MODEL SETUP

c0debreaker, 2021-03-16:
**Describe the bug**
It happens when using MULTI settings in External RF section in MODEL SETUP of my Jumper T16. Once I switch to MULTI then start choosing FrSky, before I can get to select FrSky, ASSAN shows up and I am unable to scroll thru selection anymore. It gets stuck in ASSAN. I have to exit and change MULTI back to OFF or any settings then MULTI settings again. I'd do the same process but I have to be extremely fast and accurate to get to FrSky. If I am slow, ASSAN will show up.

**Expected behavior**
Should be able to scroll thru different protocols and not get stuck in Assan

**Screenshots**
https://imgur.com/a/Yr4AyuN

**General information**
 - OpenTX 2.3.11-tbs version 
 - Jumper T16

kilrah, 2021-03-16:
Likely need to update multiprotocol module firmware.

c0debreaker, 2021-03-16:
Thank you @kilrah How do I do that?

c0debreaker, 2021-03-16:
I found this, I'll watch it - https://www.youtube.com/watch?v=8fwU220sVKc

c0debreaker, 2021-03-16:
I just finished updating it and it said on screen "FLASH SUCCESSFUL". I tried the process above again, it is not fixed. The issue still remains. Let me know if I should share a video.

kilrah, 2021-03-16:
Try our builds instead of the TBS one.

c0debreaker, 2021-03-16:
Looking at my Downloads folder, the file I used back in January 27 is `opentx-t16-lua-noheli-en-2.3.11-otx.bin`.

Which one are you referring to?

kilrah, 2021-03-17:
Your initial post says "OpenTX 2.3.11-tbs version" so you have the special TBS build, download our version from companion.

c0debreaker, 2021-03-17:
Ah, I see it on my screenshot too. Will I have to reconfigure everything again?

kilrah, 2021-03-17:
No.

c0debreaker, 2021-03-17:
Thank you. Everything is good now. The bug that was present in the TBS version didn't exist in your version. It also connects to the frsky receiver really quick now. Companion is cool. I used the write config to radio. It worked but there was no message. I think a message window must pop up so that users can be informed that writing to the radio was successful. I only knew that it worked when I rebooted my Jumper T16.
It's all good! Thanks a lot! 💪

c0debreaker, 2021-03-17:
Oh maybe the reason I used TBS firmware was because of crsf shot. I'll check it later since it's 3:40am. I'll hit the bed now.

c0debreaker, 2021-03-17:
How do I get crsf shot added with native OpenTX version?

kilrah, 2021-03-17:
You wait until it's included.

c0debreaker, 2021-03-17:
Got it. Is there an existing pull request?

c0debreaker, 2021-03-18:
Please disregard my question. I checked https://github.com/opentx/opentx/pulls and looks like there is no existing pull request for crsfshot. My guess it might take a long time before this feature support gets merged into native OpenTX.

closed 2021-03-18 by c0debreaker

kilrah, 2021-03-18:
It is planned to be included in 2.4, but yes that may be far away.

c0debreaker, 2021-03-20:
I honestly thought I won't be able to fly without crsfshot. However, I flew it today and didn't have issues at all. My quad was still very responsive. I didn't notice any latency. 😊