opentx/opentx #8381: SW setting the 6Pos

Enggmaug, 2021-03-18:
Hello, 

I'd love to be able to set the 6 position switch to a given value through a special function on my TX16S.

I understand this would not be possible for radios that have a mechanical switch, but T16 and T18 have a key for each position, that is symbolized with a LED.

I wanted to get my hands dirty and see how to do that myself, but thought I would first discuss this here. 

Any idea what it would take ? I'm a bit lost after a couple of hours browsing the code.

eng

3djc, 2021-03-18:
Otx does have no control about that, and cannot have it, as it is managed by a specific external circuit

closed 2021-03-18 by 3djc

Enggmaug, 2021-03-18:
this explains that...

thanks !

pfeerick, 2021-03-20:
Bugger! So (just to confirm) the indicator lit is managed by the hardware, not software controllable?

Enggmaug, 2021-03-20:
That's also my understanding. Is there an HW-SW interface description document anywhere available ?

kilrah, 2021-03-20:
There's a separate MCU on a board that puts out an analog voltage to emulate a pot. Low tech.