opentx/opentx #8387: Synchronize SD - wrong labeling

bigtwin, 2021-03-20:
When completing the Dialogbox for "_Synchronize SD_" at first one is asked to provide "_Local Folder_" and "_Radio Folder_". When completing the next line "_Sync. Direction_" it is referred to as "source folder" and "destination folder".

This is no good style in terms of usability and it is one of those little piece which makes it more difficult to build trust into Companion. Surley someone can figure out what is ment by those lables but it is certainly more easy and better style to have the naming consistent.

Thx for taking care of OTX!

elecpower, 2021-03-21:
Duplicate of #8319

closed 2021-03-21 by elecpower

bigtwin, 2021-03-21:
Sorry, I missed to check before starting this issue. Thx for handling!