opentx/opentx #8389: Block trim value changes with a special function.

loetefix, 2021-03-21:
It would be nice if there was the possibility to block the changing of the trim values, preferably with special functions.
The background to the story is that I use the trim values as a channel that can be changed in flight and it would be nice if i can block the changing of the value by a switch.

elecpower, 2021-03-26:
@loetefix assume you wish to hold the last trim value for the channel before the switch is true

loetefix, 2021-03-26:
i want a switch that disable the trimswitch .so if the switch is enable you are unable to change the trimvalues. not disable the trimvalue itself.for example i use the throttle trim as a source for an own channel to control rotor rpm. every flightmode have its own trim value "bank". So i can adjust the throttle while flying. when the setting are ready, i want do protect the trim from accidental adjustments with a switch.

kilrah, 2021-03-29:
Use a global variable then, you can have it different per flight mode, adjust it with trim switches, and once done remove the "adjusting" SF to essentially lock it.

closed 2021-03-29 by kilrah