opentx/opentx #8392: BUG? SD Logs not logging state of switches (SA etc)

wx4cb, 2021-03-24:
so having SD logs switched on when the aircraft is armed (via special function "SD Logs 0.25"), i've noticed that in the actual logfile that is produced, the state of the switches are not logged,. they're either -1 or 1 and that way all the way through the log file.

SF is my prearm switch so i'm expecting that to be one (as that's the one that triggers a telemetry reset, and SD logs to start running), but then SB and SC should change state just after that - SB is my "actual" arming switch and SC is launch mode.

The process then should be, SF then SC then SB, so i am expecting SF to be one and then SC and SB to go to 1 a few lines down.

I have put two grabs from the same log below which should give you an idea

[image]

[image]

kilrah, 2021-03-24:
What radio? What firmware version? Post the unmodified csv too.

wx4cb, 2021-03-24:
version is 2.3.11-otx (4328055e) dated 1.8.2021
Radio is a TX16S

[Mini_Talon-2021-03-10 .csv.txt](https://github.com/opentx/opentx/files/6200890/Mini_Talon-2021-03-10.csv.txt)

wx4cb, 2021-03-24:
@kilrah i never noticed it before, but i'm playing with dashware and wanted to put like arm state type of indicators

kilrah, 2021-03-24:
They all change between -1/0/1 as expected.

[image]

closed 2021-03-24 by kilrah

wx4cb, 2021-03-24:
@kilrah weird... so qould -1 be the equivalent of -100 and 1 is +100 right ?

kilrah, 2021-03-24:
Switches are only up/middle/down and that's represented as -1/0/1, there's no 100 involved.