opentx/opentx #8393: Different languages for the simulator FW

ProfDrYoMan, 2021-03-25:
**Is your feature request related to a problem? Please describe.**
The menu language in the simulator TX is always in englisch.

**Describe the solution you'd like**
An option to change the menu language on the simulated TX.

**Describe alternatives you've considered**
There are no alternatives.

**Additional context**
A description on how to compile on my own would be also ok (for me).

3djc, 2021-03-25:
https://github.com/opentx/opentx/wiki/Compiling-OpenTx

ProfDrYoMan, 2021-03-25:
Ok, so compiling on my own needs more steps that I hoped.

If there is an automated build setup it would be great to see multiple languages compiled for the simulator.

closed 2022-04-17 by stale[bot]