opentx/opentx #8398: Add DSM Forward Programming support for Spektrum AR630

Schaezard, 2021-03-26:
**Is your feature request related to a problem? Please describe.**
When I try to use the latest DSM Forward Programming Script with my Spektrum AR630 receiver, the configuration for failsafe etc. works fine. However the AS3X first time setup shows no text in the menu. Pressing the 'Next' Button does not seem to actually do anything.

**Describe the solution you'd like**
It seems to me like the Script is simply missing the correct translations for the return codes, the receiver is sending. So adding those should allow me to interact with the AS3X setup properly.

**Additional context**

[image]
[image]
[image]

3djc, 2021-03-26:
You should direct this to multimodule, as OpenTX does not deal with this itself

closed 2021-03-26 by 3djc

Schaezard, 2021-03-26:
> You should direct this to multimodule, as OpenTX does not deal with this itself

My bad, I planned to do that and accidentally ended on the wrong github tab it seems ^^