opentx/opentx #8401: Radiomaster TX16S Hall Gimbals drifting with temperature change

Seba112PL, 2021-03-29:
Video showing problem (calibrated at 25 degrees celsius outside around 2)
https://youtu.be/YKdf8Qok9bw

Steps to reproduce:
1. Make sure that you dont have any curves in your radio (stock OPENTX from RadioMaster have that curve at the center)
[image]
2. Calibrate your radio at room temperature.
3. Go outside (winter time -5 degrees etc) or put your radio inside a fridge for abour 30 minutes.

Your stick position should start drift away from the center to one side.
After you go back to room temperature your stick will drift back to center or if you calibrate your radio outside and then go inside your stick will start to drift away from center to other direction.

Solution (?)

Use inside temperature sensor/attatch UART temperature sensor to monitor temperature and correct hall readings.
(Create additional calibrating procedure that will "fight" temperature changes
1. Calibrate your radio in room temperature
2. Put your radio stick to center
3. Put your radio inside somewhere cold (fridge (?)) for some time
4. End calibration
5. Use that data to compensate for temperature effects on sensor readings

richardclli, 2021-04-02:
I think change the Hall Sensor with a better chip is easier.  There are many chips out there that contains circuitry to ensure the output stable across wide ambient temperature range.  However, the chip used in Jumper/Radiomaster is not this type.

I am trying to install a better chip: Allegro A1391 series (Frsky used this) to my Jumper T16 and see if it can help to improve the quality of the signal. ^.^

Seba112PL, 2021-04-02:
Its really a shame that Radiomaster and Jumper cheaped out on the most important part of a radio... gimbals

Anyway. Took radio apart and Hall sensors are on a seperate board that can be easily removed. Replacing hall sensor should be easy to do for someone that have some SMD soldering experience. Thank you for providing a part number for halls that frsky used

Please let me know how this mod went for you and if the problem is gone after hall replacement.

drtrigon, 2021-05-02:
Is there an internal temp sensor on TX16S available?

Would that allow for a rough correction?

3djc, 2021-05-02:
Nothing we can do on otx side, so closing this

closed 2021-05-02 by 3djc

Hasi123, 2022-01-13:
This issue is also present on multiple X10S radios. Most MCUs do have a builtin temp sensor, so a software calibration is (in theory) possible.

rotorman, 2022-01-13:
Indeed STM32F429BIT6 in TX16S has an internal temperature sensor and one can easily read it out. But the MCU core is typically much hotter than the ambient and this is also what the sensor outputs.

Here an example measurement done on TX16S in a temperature chamber, where the green line is the output of the core temperature sensor and the orange line is the output of the external reference sensor (LSM6DS33 in this case):

[image]

It is not as easy though as taking the core temp. value and subtracting an offset from it, as during cold start the core temp. sensor outputs with no or less offset, as it has not yet warmed up.

drtrigon, 2022-01-14:
Basically there is a good correlation. Can you plot the difference between the 2 curves? Can you post the raw data?

Could you do a "calibration" that relates the 2 temperatures using the chamber that can then be tested by other users?

drtrigon, 2022-01-14:
Do the hallsensors have a temperature output? Do they have any other value that could be used as reference?

rotorman, 2022-01-14:
You can find the raw data at: https://github.com/rotorman/TX16S_HallDriftComp/tree/main/tests
The Allegro A1392 hall-effect sensors on TX16S gimbals do not output any temperature and in whole radio there is no other component that provides w/o hw-mods temperature value to the main STM32F4.

Hasi123, 2022-01-17:
@rotorman Wow, that is some extensive reasearch you have done on the subject. At the end did you use your data to implement some sort of compensation in software?
It doesn´t really matter if the MCU temp sensor has an offset, or a sligtly different curve as long as you use the sensor for calibration and the compensation.

rotorman, 2022-01-17:
My original experiments with RM TX16S stock hall gimbals started with this exact idea of being able to minimize their drift in firmware.

A large part of the temperature drift on stock RM TX16S gimbals stems from the too high value input resistor of A1392 VREF pin. By replacing this, against a lower value (stock 100 kOhm, against 10 Ohm), a significant boost in temperature stability can already be achieved.

Further improvement is possible, but not universally possible without calibrating each radio separately IMO, as my the experiments showed. The issue here lies in the semi-random orientation of the magnets. On a positive note, on V3 hall and AG01 gimbals both issues are remedied (VREF resistor value and magnet orientation).