opentx/opentx #8402: Rename issue SD CARD menu on X9D+ unable to shorten names of files and folders

alm-nl, 2021-03-29:
I have discovered that in the SD CARD menu it is not possible to rename files and folders to a shorter name than the one it already has. The space character does not appear when browsing through the characters so it is not possible to go from name123 to name12. It is possible to make it longer, but not shorter.
It would be very nice if this would work for those situations that you don't have a pc at hand while out flying or for those who do everything from the transmitter itself.

I enter through the character positions with ENT and use + and - keys to go through the available characters, but space (' ') is missing.

 - OpenTX version 2.3.11 (official release version)
 - Frsky X9D+

It was confirmed by Miami Mike on rcgroups. Risto confirms it does work on his TX16S, so it might just be a B&W radio issue.

Thank you.

Darko-FC, 2021-03-31:
Hopefully this is the right place for similar/related issue.

 Can you please provide consistent behaviour for renaming files and folder names on (all) radios ?

alm-nl has provided one example - and I have recently found similar problem - while trying to rename log file name on SD card. If I tried to change just few characters and exit - new file name is not saved.
Same is working fine for model name - I can change just one character for example and exit - and new model name is saved.
(and that is my current workaround for log file name change - I actually change model name for new log file name (same day))

I don't have idea about work needed and realize it is more nuisance than functional issue (not high in priority) - but it would be nice if you can find time to look this. Thanks.

mpjf01, 2021-04-01:
Changing the model name is the workaround suggested in #7752.

dgatf, 2021-04-01:
> while trying to rename log file name on SD card. If I tried to change just few characters and exit - new file name is not saved.

You have to long press enter at the end (or over any space character) to save the new file name. Exit aborts renaming

alm-nl, 2021-04-01:
Nice find, hope it will be accepted for 2.3.12 release.

Darko-FC, 2021-04-02:
mpjf01 - thank you for that reminder.
dgatf - thank you very much for "over any **space** **character**" part / solution.

alm-nl, 2021-04-02:
Tested the fix by @dgatf and works perfectly! See also his submission as I responded to it.

Names can be made longer and shorter using this fix, so perfect! 😎👍👍

PR #8405 (2021-04-05): Add space to SD edit name
Fixes #8402. Only affects B&W radios

This PR allows up to the space character in SD rename file for B&W radios. This allows to shorten the file name.
closed 2021-04-05 by bsongis