opentx/opentx #8404: Vario programming under Open-TX

Basti77, 2021-03-30:
Servus,
maybe someone can help me.
I am trying to set up the vario of the g-rx8 receiver in the transmitter (XLite, current openTX). 
I get it to announce altitude to me on/above a switch, but not the classic beep beep with the change in frequency to delta V.
The option about special function on a switch "Vario" generates only constant beep beep.

kilrah, 2021-03-30:
They are here and Source needs to be set to Vspd.

[image]

closed 2021-03-30 by kilrah

Basti77, 2021-04-01:
thank you very much. 
it works now.
I think I have not found the place because the numbering of the menus is not correct. 
So far I have not missed anything, but what can be missing with me so that with me 12 is 13.
[image]