opentx/opentx #8407: script error

kiwi41000, 2021-04-02:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
operating a frsky taranis x9d+ 2019  opentx 2.3.11 While trying to use the frsky sbec function I'm getting Script syntax error attempt to compare number with nil. Dan

elecpower, 2021-04-06:
@kiwi41000 I took the liberty to remove the template text from your issue.

elecpower, 2021-04-06:
@kiwi41000 does the script crash almost immediately or does it display information or menu and crash after a selection?

kiwi41000, 2021-04-06:
Upon opening of the program for the SBec voltage adjustment Lua script it fails immediately

kilrah, 2021-04-07:
Make sure you get the current script version, there have been old broken ones around

kiwi41000, 2021-04-07:
This is with the Tranus 2019 in D 16 mode to an x receiver and I have re-loaded the software three or four times just to make sure it’s correct. Might it work if I were using access protocol receiver.

3djc, 2021-04-07:
what esc ?

kiwi41000, 2021-04-07:
Neuron 80 to a s8r but it works great with the Taranis x9d

kiwi41000, 2021-04-07:
I have also updated the firmware several times and no good

kiwi41000, 2021-04-07:
Neuron 80 to a s8r but works good with the Taranis x9d

kiwi41000, 2021-04-07:
Neuron 80 but works fine with the Taranis x9d

kiwi41000, 2021-04-17:
Sorry if you need anymore information Ive been having problems with my email curious if you’re making any progress

closed 2022-04-16 by stale[bot]