opentx/opentx #8409: Radiomaster TX12: sbus as input for trainer/master in JR module bay not working

Riffer, 2021-04-02:
Using sbus receiver as input device for does not work

Steps to reproduce the behavior:
1. Connect an already bound (to another TX) standard SBus-Receiver into JR Bay (beginning with the second pin from the bottom, GND, VCC, SIGNAL)
2. Power up TX12
4. Select Master/SBus in model settings (power for RX turns on)
5. Map incoming channels TR1 to TR8 to input and mixer
6. Show channels in the front end on TX12

**Expected behavior**
While moving trainer sticks on X9D plus the TX12 channels should show corresponding movements. 

**General information**
 - OpenTX version 2.3.11 and N470
 - Hardware TX12 (and X9D plus as sending device)
 - I ensured that this szenario works the other way around, where I used my X9D Plus as master/sbus and the TX12 as a trainer device.

Riffer, 2021-04-25:
Is my expectation that I can use SBus as input in the JR bay wrong?

3djc, 2021-04-26:
Investigating your issue I stumbled into another one, which is that synced pulse has broken SBUS trainer...

3djc, 2021-04-28:
Ok, it turns out there was a software issue affecting nightlies that is being fixed.  It also appears there is a hardware design flaw on TX12 that prevent SBUS trainer to work. I found a fairly simple hardware fix, I'm talking with RM so they confirm (or not) my findings

3djc, 2021-04-28:
Here is what needs to be done for those who really need trainer from ext bay working (cppm or sbus)

[image]

[image]

closed 2021-04-28 by 3djc

Riffer, 2021-05-02:
[image]

I guess I have a new problem and I do not know whom to ask, but you: The mainboard of my TX12 has "TX12S" on it, and comparing it with your picture I fail. I am not experienced in SMD electronics but I guess that item on the left is a capacitor while the right one is a resistor. The right group looks like the same as yours but the item on the left is not only much bigger but also a little bit moved. 

I hoped the update alone could help, but I guess even with this 'never' version of the board struggles with SBus - so I think I need to solder the wire. But where... ?

3djc, 2021-05-02:
I have asked Radiomaster to supply the information about the right points for this mainboard

3djc, 2021-05-06:
Not the best picture ever, but here you go !
[image]

georgweb, 2022-01-03:
[image]
Done that, still no trainer function. I'm measuring approx. 10kΩ both ways on that tiny element on the right end of the cable - so I'm confident that my soldering worked out OK. Been using trainer function via XM+ rx in module bay of QX7s so I am also confident that I have indeed triple checked everything else. TX12s is on otx2.3.13.

Riffer, 2022-01-04:
I am sorry for my late response, but must agree, that mod didn't work for me either.

leofelipelf, 2023-04-07:
> [image]
> 
> I guess I have a new problem and I do not know whom to ask, but you: The mainboard of my TX12 has "TX12S" on it, and comparing it with your picture I fail. I am not experienced in SMD electronics but I guess that item on the left is a capacitor while the right one is a resistor. The right group looks like the same as yours but the item on the left is not only much bigger but also a little bit moved.
> 
> I hoped the update alone could help, but I guess even with this 'newer' version of the board struggles with SBus - so I think I need to solder the wire. But where... ?

leofelipelf, 2023-04-07:
> [image]
>
> I guess I have a new problem and I do not know whom to ask, but you: The mainboard of my TX12 has "TX12S" on it, and comparing it with your picture I fail. I am not experienced in SMD [...]

Hi, Riffer! Sorry comment in this post. But i really need help to identify a component. Could you help me please?
[image]

Riffer, 2023-04-07:
I am sorry, can't help.

Maybe @3djc ?

leofelipelf, 2023-04-07:
> I am sorry, can't help.
> 
> Maybe @3djc ?

Thanks, Riffer!
It's a try. Hope i could recovery. I just connect in the wrong voltage.
Look at this 🤦🏻‍♂️
[image]

sadovenko, 2023-06-12:
> Hi, Riffer! Sorry comment in this post. But i really need help to identify a component. Could you help me please? [image]

Hello
Did you have a chance to identify the component?

leofelipelf, 2023-06-12:
Hey!! Yes, the part number is CS5090E. In my case, i replaced it and it worked, hope it works for you too.

rotorman, 2023-06-12:
CS5090E is a 2-cell Li-Ion/Li-Po charge controller IC. Has only to do with charging the radio battery and is not connected to main radio microcontroller and cannot influence or be influenced by the firmware of the radio.

sadovenko, 2023-06-12:
Thank you very much 

Sent from my iPhone

> On 12 Jun 2023, at 13:51, Leonel Felipe Lima Figueiredo ***@***.***> wrote:
> 
> Hey!! Yes, the part number is CS5090E. In my case, i replaced it and it worked, hope it works for you too.