opentx/opentx #8410: Radiomaster TX12: not all system screens respect contrast an font settings

Riffer, 2021-04-02:
Some screens at TX12 do not obey the contrast settings (and maybe font as well)

**To Reproduce**
Steps to reproduce the behavior:
1. Start system screens
2. Page through them

**Expected behavior**
All contrast and font should stay the same

**Screenshots**
[image]
[image]
[image]

**General information**
 - OpenTX version 2.3.11 and N470 with option sqt5font (DE)
 - Hardware Radiomaster TX12

3djc, 2021-04-02:
Nothing we can do about, hardware behavior

closed 2021-04-02 by 3djc

Riffer, 2021-04-02:
Maybe my initial explanation was not clear enough.

All other screens are okay and only 2 single ones are caused by hardware behaviour? Not plausible to me.

If it happened randomly I might understand your statement, but that is not the case here. Why should the light or contrast change at all, when you page through any information at all? There must be code that changes the brightness or contrast values of the display.

Riffer, 2021-04-03:
To show it more detailed I turned up the overall contrast - in this GIF you can see, that screens 4 and 7 are totally over contrast. 

[image]

Here is a link to the original: https://photos.app.goo.gl/z8kLyKYA12tfaEyT8

@3djc

kilrah, 2021-04-03:
The LCD just has overall contrast that depends on what content is displayed, nothing we can do about that.

Riffer, 2021-04-04:
@kilrah Thank you for the detail - so - if there were more text at the display the contrast wouldn't freak out? I will give it a try...