opentx/opentx #8413: Radioking tx18s opentx support

marcus202, 2021-04-03:
Hello, I am a bit confused. So I bought this radio (Radioking tx18s) thinking it had the same opentx experience as all of the other ones, but it turns out that I can't update it because there are no official updates for this radio. It uses a custom opentx version, sadly it wasn't marked anywhere or I missed it because I didn't know what I was supposed to be looking for. Okay I'm new to this, but e.g. I want to set the flexr9m to on but since there is no official firmware for it, is it even possible?

Thank you in advance for answers:)

3djc, 2021-04-03:
This radio is not a supported OpenTX radio (https://www.open-tx.org/radios)

closed 2021-04-03 by 3djc

marcus202, 2021-04-03:
Yeah, I know that now, but that's why I'm asking. Is it even possible to enable flexr9m on it?

kilrah, 2021-04-03:
Believe it's basically identical to the T18, firmware for that might work.

marcus202, 2021-04-03:
Yeah, I tried it but for some reason, my sticks wouldn't move.

kilrah, 2021-04-03:
Then you should probably try to see if someone in the community found something, or contact the seller/manufacturer.
They never contacted us so we have no idea about that thing.

marcus202, 2021-04-03:
Yep i already asked them:)

rotorman, 2022-05-10:
@Nich2245 as RadioKing TX18S is not a supported radio (not by OpenTX, nor by EdgeTX), you are a bit in a pickle with your radio choice.
Anyhow, as the mainboard of TX18S should be the same, or mostly the same as Jumper T18, you can try to flash Jumper T18 firmware to it in DFU mode.
For this, download latest OpenTX Companion and install it on your compture.
Open Companion and set up a profile for Jumper T18.
Power off your TX18S and hook it up via USB to the computer with companion - this will connect the radio in DFU mode that is required to fully flash the firmware of the radio in one go (bootloader and the firmware part).
Now try to download and flash the firmware via OpenTX Companion.
The flashing via DFU mode takes significant time (numerous minutes) - be patient. After flashing finishes, reboot the radio and put OpenTX 2.3.15 SD card content of horus to the SD card.
If all went smoothly, and your radio hw is not faulty, you should be greeted with a running system again.

If you have further questions, GitHub is a wrong place for this kind of non-developmental support - Discord would be better.