opentx/opentx #8414: Stick data updates much slower than RF packet rate

Faultyclubs, 2021-04-03:
I am using the Radiomaster TX16S with Redpine protocol (2ms).

Using Betaflight Blackbox debug logging I can see that the RF packets are indeed arriving at 2ms as expected, HOWEVER the underlying stick data is only be updated every 6-8ms.   Meaning every three or four packets contain the same data.  This duplicated data is a problem, worse even that dropped packets, and especially affects feedforward causing rough motors.   And obviously negates the latency potential of the 2ms Redpine protocol.

I asked Radiomaster and understood from them the TX16S runs at 400k baud internally and should be fast enough therefore it is a software problem.

Is this a known problem and if so what  is causing the problem please and can it be fixed?

Thanks.

kilrah, 2021-04-03:
The protocol used by the multimodule isn't designed with speed in mind but flexibility.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

Faultyclubs, 2021-04-03:
What is the maximum RF packet rate that works properly then, please?

Also the link shows a "fast serial" interface to OpenTX.  My simple minded calculations shows that 400k is fast enough.  Isn't it?   Is OpenTX not using the full potential of the TX16S?   I'm sorry but am not quite understanding where the bottleneck is.

kilrah, 2021-04-03:
The software running on the RF module needs to be talked to at 100kbps and a frame is around 4ms. OpenTX just provides what the module wants. 

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/931ba2bd68f345f1c6eb59f6f0bb64e25e484bc5/Multiprotocol/Multiprotocol.h#L801

Faultyclubs, 2021-04-04:
I see.  But is that a legacy kind of thing?  If the TX16S is indeed running at 400k interface to it's internal MPM then all that is needed would be software updates (coordinated on both sides) as Radiomaster suggested...correct?  I realize that would be pushing the envelope and require work but seems worth it to get properly working protocols.   If it's even technically feasible of course, I can't find any hardware information on the TX16S to confirm that it really does support 400k to the MPM.

Also, not trying to be argumentative but are you sure about the 4ms?  That is not consistent with what I'm seeing.  It seems to be much higher.  I can show the logs if necessary.   At 4ms I would switch to S-FHSS to get consistent packets but even my 6.8ms S-FHSS seems to be duplicating packets.  I'm often seeing 8ms on Redpine.   I am running v2.3.9 though in case there's been a recent change.

dgatf, 2021-04-04:
> Using Betaflight Blackbox debug logging I can see that the RF packets are indeed arriving at 2ms as expected, HOWEVER the underlying stick data is only be updated every 6-8ms.

That perfectly aligns with the period defined to update the mixer for the Multimodule in Opentx:

https://github.com/opentx/opentx/blob/a167ded9b534f373904e4424e053187b90237b8a/radio/src/pulses/pulses.h#L233

Maybe reducing this parameter improves the latency. I don't have radio now to test though

Faultyclubs, 2021-04-04:
Wow, great find.  I wish I knew my way around the code like some of you guys do.

Yes, 7ms explains what I'm seeing on Redpine.   It also explains why S-FHSS is getting buggered up.  S-FHSS is 6.8ms protocol which just misses the cutoff.  Can we at least adjust this parameter so S-FHSS works properly (ie consistent 6.8ms with no stale stick data).

Ideally we improve Redpine too.  Constant 6ms is better than jittery 6-8ms.  And of course 4ms is better still...

I can't help on the development but can test changes.

3djc, 2021-04-04:
That’s not how it works, the multimodule ask for a specific rate as it sees fit

raphaelcoeffic, 2021-04-06:
@Faultyclubs if 500Hz is what you want, you should really have a look at ExpressLRS, because it's working there  ;-) (together with the new sync merged yesterday into 2.3)

closed 2021-04-06 by raphaelcoeffic

Faultyclubs, 2021-04-06:
Close this if you want but, with all due respect, you guys didn't do anything. :)

S-FHSS is a well established protocol and should be fixed IMHO. The only stated technical limitation is the 100k link which is more than sufficient for S-FHSS to work properly.

In the process Redpine might be improved too since 100k should support better than 8ms.

ExpressLRS does sound promising.  Especially if it works on microwhoops with SPI receivers.  I had heard that wasn't going to happen though.  If not, then perhaps Redpine could be ported onto the ExpressLRS hardware for the tiny whoop guys.  I think the transceivers are capable.

kilrah, 2021-04-06:
> but, with all due respect, you guys didn't do anything. :)

Of course, because as mentioned there's nothing for us to do at this point...

If you want 2ms redpine then first find a module that can run 2ms redpine, then we can see to make it work from opentx :)

raphaelcoeffic, 2021-04-07:
> Close this if you want but, with all due respect, you guys didn't do anything. :)

@Faultyclubs

For my part, I strongly disagree:
- coordinated with 3 different parties / companies
- added a new sync mechanism
- added dynamic rates to adapt to the module's wishes

I cannot possibly go as well to each of the module makers and implement it for them. However, there is now a common interface for modules to request sync and dynamic rates. I'd say, now it's up to you ;-)

brycedjohnson, 2021-05-05:
@Faultyclubs agree with the above guys.  I wrote redpine, it was designed around deviationTX and later ported to MPM.  Currently it is MPM limitations prevent the higher updates.  I did some brief working looking to improve that, but haven't spent time on it lately.  It certainly is possible though, just need someone with some time to dig into it!

On a side note: SX1280 chipset on ELRS talking redpine to cc2500 spi receiver is on my list of things to try at some point.  Looks theoretically possible, but not tried out yet.

raphaelcoeffic, 2021-05-06:
> On a side note: SX1280 chipset on ELRS talking redpine to cc2500 spi receiver is on my list of things to try at some point. Looks theoretically possible, but not tried out yet.

@brycedjohnson that would be great! and if we could do the same thing with Frsky D8/D16, it would be just perfect!

brycedjohnson, 2021-05-06:
@raphaelcoeffic D8 D16 might not be possible because SX1280 only supported a few bitrates in FSK mode.  B8 D16 use an odd ball (?)46Kbs rate.  Redpine fast mode is 250Kbs that look like one of the supported rates.

[image]