opentx/opentx #8418: Deleting an input in companion "destroys" model

ProfDrYoMan, 2021-04-04:
**Describe the bug**
Deleting an input in companion "destroys" model

**To Reproduce**
See attached model file.

Steps to reproduce the behavior:

Before:
[image]

Deleting I8:
[image]

Result:
[image]

Even L18 is gone completely!

**Expected behavior**
Deleting the input but nothing more.

**Screenshots**
See above

**General information**
 - OpenTX version 2.3.11
 - Hardware TX16s
 - OpenTX Companion 2.3.11 Windows

[Bug.zip](https://github.com/opentx/opentx/files/6255021/Bug.zip)

elecpower, 2021-04-04:
In an attempt to maintain the many possible cross references, deleting an input, mix, etc can affect other configuration that refers to the deleted input, mix, etc or the effect of shuffling up the items below. This can trigger a cascade of updates throughout the model.
Without seeing your whole configuration for this model I cannot tell if this is working by design or you have found a bug.

ProfDrYoMan, 2021-04-05:
That is why I added the otx. See last line of the report. Bug.zip.

elecpower, 2021-04-05:
Apologies missed it. I'll take a look when I have some time over the next couple of days.

elecpower, 2021-04-05:
Found the cause of your issue. The screenshot of your Logical Switches shows L18 referencing L11 which has no V1 activation switch.
[image]
The model reference update iteratively tries to remove inactive/incomplete configuration because of the Delete action. However it can also remove pre-existing inactive/incomplete configuration.
In your case L11 is deleted as it has no activation switch in V1 which removes the V1 L11 switch from L18 (and everywhere else it is referenced) which causes L18 to be deleted and the switch in I11 to be removed.
Will need to review the logic to determine if, due to the iterative nature of the update, it is practical/possible to differentiate the cause of inactive/incomplete configuration.

ProfDrYoMan, 2021-04-05:
Will look into my config there. It is possible that I missed something in L11. :)

ProfDrYoMan, 2021-04-05:
Could remove L11 and trigger L18 with Act.

PR #8429 (2021-04-06): Model update references improvements
Fixes #8418 and Fixes #8428
Set 'new' flight modes in inputs and mixes as a result of deleting/inserting flight modes to inactive which is a safer option
closed 2021-04-06 by bsongis