opentx/opentx #8419: Horus X10 express crashes when MPM overflows it with telemetry sensors

notlebowski, 2021-04-04:
I'm writing this issue on behalf of a friend who recently bought a Horus X10 express after his Horus X10 let him down structurally.

He's not having this problem on his Horus X10S which is running OpenTX 2.3.9
His Horus X10 Express falls into emergency mode if he selects the DSMX-protocol on his MPM running version 1.3.2.61 and is discovering sensors.
It keeps on discovering bogus sensors until it crashes.

He sent me a video which I attached

notlebowski, 2021-04-05:
My friend still has this problem.
To make sure it is not a problem with his settings he formatted the SD-Card, copied a freshly downloaded SD-Card content on it and turned it on.
The radio then goes through the motions having to calibrate and so on.

Then he created a new model, bound it to a receiver and again his telemetry started to fill up with bogus sensors.
Clearly not how it should be.
He doesn't have any genuine Frsky receivers, so he can't test those.

He repeated the process with his old X10S (not an Express) using the same module and doesn't have this problem

He would really like to know if something is wrong with just his X10 Express or that there are currently some issues with OpenTX regarding the X10 express....

richardclli, 2021-04-06:
Multiprotocol 1.3.2.61 requires OpenTX 2.3.10, upgrade to 2,.3.10 and the problem will gone.

notlebowski, 2021-04-06:
He's running 2.3.11 on his Express (_I now noticed I left that out_)
I believe a 2.3.9 does not even exist (_at least that's what he said to me_)
He's running 2.3.9 on his old Horus X10S which doesn't have that problem with the same MPM

Devil-67, 2021-04-18:
My X9D + SE 2019 with 2.3.10 also finds nonexistent sensors but with Frsky receivers G-RX8 and GR-8, but I cannot confirm the emergency mode with me.
I'm not sure if it's an OpenTX or a Frsky problem.

notlebowski, 2021-05-10:
My friend has found a fix in modifying the MPM (it also did it with my MPM).
Apparently the signal it's giving is too weak for the newer Horus X10 express or there is just more noise.

He needed to bridge 2 contacts with a wire and his problem was gone.
The Horus X10 express was returned to Engel, but they were unable to reproduce the problem. That's not that strange because they didn't at all repeat the scenario in which he was having this problem.

They like to have the benefits of the sales that come with this brand in combination with opensource projects like the MPM, but when a problem arises they suddenly "go official" and tell you to flash special OpenTX-versions without MPM support and only use "official" Frsky-products.

At least his problem was finally fixed thanks to the efforts of Pascal (MPM) and someone else.

Most of the info I'm having comes from this friend. I haven't had any direct dealings with any of the involved beside my friend.

This ticket could be closed as it is hardware related.
On the other hand it could be left open as OpenTX shouldn't crash because they are getting noise on the telemetry and subsequently crash the model.

I leave it to the devs.
I

closed 2022-04-16 by stale[bot]