opentx/opentx #8420: Multiprotocol for 2.1

Pressfit, 2021-04-05:
I still use a th9x and bought me a multiprotocol unit. The only problem is its not implemented in fw2.1 and fw 2.2 do not support my radio. By trying to implement it by myself, i even failed to compile the unmodified code due to python errors.
Is it possible to get a early stage of opentx without python wich i can compile in microchip studio or in winavr. Or any kind of documentation to write it in C/C++ by myself?

greetings from germany

pfeerick, 2021-04-06:
I don't know if you'll have much luck, but this may unfortunately be a case where it would have been better to get a MultiProtocol module that supports PPM mode (as opposed to being a serial only one), i.e. the iRangeX IRX4 Plus ... https://www.banggood.com/IRangeX-IRX4-Plus-2_4G-CC2500-NRF24L01-A7105-CYRF6936-4-IN-1-Multiprotocol-STM32-TX-Module-With-Case-p-1225080.html ... this version lets you set the protocol via a switch on the back, as well as having a bind button.

raphaelcoeffic, 2021-04-06:
Seriously, you should consider upgrade your radio, so you can make use of the newer firmwares. Otherwise, you will end up with a huge load of work, and no support....

closed 2021-04-06 by raphaelcoeffic

pfeerick, 2021-04-06:
AFAIK, the FlySky TH9X is almost, if not identical to the Turnigy 9x, which has a comfortable feel and well tested hardware, so if you want a solid radio without fancy features like voice, colour screen, etc, it's still a great one. Whilst OpenTX has moved on from AVR based transmitters, Er9x added a lot of features like multimodule support, etc, making it worth investigating if you want to get a more up to date firmware for your transmitter. Mike still appears to be keeping up support, as recently as four months ago. You will still need to find out if the module will just work with your transmitter, or if a hardware modification would be required.

kilrah, 2021-04-06:
Mike just discontinued the 9X as well a couple of weeks ago, but yes er9x has got more feature for it than we do. Might be a better choice if you want to use that radio.

pfeerick, 2021-04-06:
Damn, just saw that - wish there had been some mention on the er9x site or github! :frowning: Thanks for the update! So no new features for the 9x, only bugfixes. Entirely understandable since it was essentially a one man show. 

https://www.openrcforums.com/forum/viewtopic.php?p=153461#p153461

Pressfit, 2021-04-07:
thanks to all for the replies. 
I actually soldered a button and a 4 bit dipswitch on it, like in the link from pfeerick. It works but its uncomfortable. 
Didn't know that er9x is more up to date. I will try. 
I think its not so difficult to include the multiprotocol into, i just can't compile it.