opentx/opentx #8430: OpenTX takes forever to create a new folder in windows 10.

Freddan-67, 2021-04-06:
OpenTX 2.3.11 on Windows 10 Pro 64-bit "Microsoft Windows [Version 10.0.19041.867]" takes over one minute to create a folder on "C", a SSD disk in windows. Eventually the program unfreezes but it takes a very long time. 
I was creating a new folder when I was trying to save models and settings. Windows froze for a couple of minutes.
I was creating a new folder when I was downloading new firmware.
The freeze happened both times. The disk is a 70% filled 1T corsair force PM510 with no hardware issues.
While the OpenTX freezes the windows explorer seems not working correctly, I can for instance not access the SD card in the radio during the freeze.
This is of course more of an annoyance than a show stopper in priority, but never the less worth looking into.

Freddan-67, 2021-04-06:
The Radio is RadioMaster T16S with firmware 2.3.10-otx from 2020-10-06. Opts: Crossfire, ghost, flexr9m, afhds3, internalmulti, multimodule, lua,luac.

elecpower, 2021-04-06:
Use Win 10 and Ubuntu 20.04 all the time running Companion and have no issues on either OS like the one you described.
There are many processes and services running on Windows that can cause the issue. Have you looked into Task Manager and/or Resource Monitor to see what is consuming resources?

3djc, 2021-04-06:
I'm with you @elecpower , OS related issue

Freddan-67, 2021-04-06:
Nothing spectacular takes up resources in task manager, everything idle. Do I have to install VisualStudio to get more info?
I got no problem to create a folder outside of companion.

kilrah, 2021-04-06:
I've had Explorer also freeze for extended periods of time on benign actions and it was caused by some program's shell extensions (dropbox or something, don't remember). Disabled all those I didn't really need and the issue was gone.

Freddan-67, 2021-04-06:
I tested on a laptop (Lenovo X1 Carbon G4/Intel i5 6300) with only USB2 hardware and then it was working great. So there may be something with the non working computer that is awkward
Ryzen 5 3600X
MSI X470 GAMING PLUS MAX
AMD USB 3.10 eXtensible Host Controller 1.10 (Microsoft)
AsMedia USB 3.1 eXtensible Host Controller 1.10 (Microsoft)
Nvidia USB 3.10 eXtensible Host Controller 1.10 (Microsoft)
Nvidia USB type-c (not tested)
The Comment from @kilrah seems to be the same for me. I simply have do debug my computer. Even without running OpenTX the computer hangs while creating a folder. Sorry for not testing enough before reporting, but I really havent seen the computer hang at creating a folder. You can close the issue, thanks for your patience with a N00B.

kilrah, 2021-04-06:
Here's a useful tool for it:
http://www.nirsoft.net/utils/shexview.html
I've disabled the Dropbox and PDF architect ones here, also TortoiseSVN which I had uninstalled long ago but for some reason it didn't unregister the handlers which could have been one of the causes for my issues...

[image]

closed 2021-04-06 by kilrah