opentx/opentx #8433: Widget/switch source problem

minzukas, 2021-04-06:
OpenTx 2.3.11
Radio Radiomaster Tx16s

Unable to select source when trying to change top bar widgets through widget settings. Roller seems to work fine in other menus, though when selecting source, always jumps to SC switch. I can select subgroup, f.e. telemetry, but when scrolling with roller it jumps to SC again. SC seems to work also fine.
Reflashed firmware - same issue. Just noticed SC up warning poping up during startup. When checking switch - I get 100, 0, -100. Opened up radio, reconnected wire harnesses, same thing...
 Any ideas how to put something like RxBt up? Disabling switch, something else. Myself I am out of ideas and patience.

P.s. sorry for previous post with false fw version.

3djc, 2021-04-06:
I'm sorry but you seem to have an hardware issue with SC, that would trigger 'autoswitch' detection. This site is to track Otx bug, which this is not. Fopr community support, please go to RM FB page, or forums like rcgroups or openrc

closed 2021-04-06 by 3djc

minzukas, 2021-04-06:
Is it fw that when SW is down roller works just one way?

3djc, 2021-04-06:
As said above, hardware issue