opentx/opentx #8436: Add ability to multiply telemetry value by constant

xoedusk, 2021-04-07:
I would like to create a Calculated Telemetry Value that is equal to an existing telemetry value multiplied or divided by a chosen constant. This can be useful, for example, for calculating average cell voltage when only the total is known. 

Eg, I want to go to the Telemetry page, create a new calculated value, choose VFAS (Voltage), and choose to divide this by 6.00.

ahmad-kassem, 2021-04-08:
I would like to have the ability to multiply and divide the sensor value by a decimal constant.

Woita2095, 2021-04-22:
Very useful feature. I miss it too. Decimal constant would be great.

arthurp, 2021-04-22:
Previous discussions and PRs related to this have recommended using the `Scale` parameter on the sensor. This can work in limited cases. The problem is that the UI field for `Scale` only has 2 decimal places, but the correct scale for sensors is often far more precise than that. The internal defaults are precise, but it's impossible to enter the correct value manually. This means its also impossible to enter a correct multiple or fraction of the scale factor to get a multiplied or divided version of the sensor. So I don't think `Scale` can address this issue.

I think there needs to be a separate "post-scaling" factor that is applied after the offset and scale that get the sensor into the correct units. That way we can use the default correction values and then rescale the final output. The easiest seems to be "Divide" and "Multiply" functions as computed sensors. Clearly people could misuse this and compute meaningless values, but that's true of the other computed sensors too (running total of voltage isn't meaningful, but can be computed).

notlebowski, 2021-04-22:
I thought this was about my ticket that I created 2 years ago.
You could have asked to re-open it.
https://github.com/opentx/opentx/issues/6293

Dweinbach, 2023-01-13:
I'd really like this feature too, for the use case of dividing the telemetry voltage by the number of cells to announce a calculated per-cell approximation.

Is there a way to upvote this?

3djc, 2023-01-13:
For this purpose, you have lua script that work great and give you direct percentages

functionpointer, 2023-01-13:
> I'd really like this feature too, for the use case of dividing the telemetry voltage by the number of cells to announce a calculated per-cell approximation.
> 
> Is there a way to upvote this?

This is specifically why I developed this feature.
With some logical switches this can even work if you fly multiple configurations, e.g. 3s and 4s with the same model.