opentx/opentx #8445: Change to white after use of INAV lua

hpiercey, 2021-04-13:
When ever I use the INAV lua scrip certain items loose their colour or turn to white. The quadrants around the off logo are white when turning off after the use of the lua. The model names on the model selection page turn white after the use of the same lua. It looks as if the names are not there but under closer scrutiny they are there but white on the white background not black as normal. If the Tx is turned off things revert to normal until the lua is selected again. Other than this I have not seen any other issues.
I'm using a Radiomaster TX16s running open tx 2.3.11-otx(4328055e)

[image]

[image]

Regards Hector Piercey

raphaelcoeffic, 2021-04-13:
@hpiercey please tell the coder who made that LUA widget that he should not modify the custom color without changing it back ;-)

closed 2021-04-13 by raphaelcoeffic

kilrah, 2021-04-13:
Pretty sure that was fixed in the script long ago, must be using an old version.