opentx/opentx #8449: Taranis QX7 won't start when connected to USB

Noahdecoco74, 2021-04-15:
I've searched a bit on the internet and the only related problem is to charge the battery through usb.

When I want to start my QX7 I have to connect the battery and disconnect the USB cable. The controller won't start with either only the USB or the USB and the battery pluged in. 
When the controller is off and I connect the USB cable, windows tells me "STM32  BOOTLOADER" is plugged in.

I would like to be able to start my controller without having to plug-in and unplug the battery every time.

My controller is a QX7 ACCST under openTX 2.3.11
[image]

3djc, 2021-04-15:
This is done in hardware, nothing that can be done in software

closed 2021-04-15 by 3djc

kilrah, 2021-04-15:
The radio will not start with USB connected, that's normal based on how the hardware is designed.