opentx/opentx #8451: S-Port-Mirror for Taranis X9D+ 19 is missing

RMFPV, 2021-04-16:
**Describe the bug**

https://github.com/opentx/opentx/wiki/Taranis-IO-ports

This section includes OpenTX serial port settings that has S-Port-Mirror option.
S-Port-Mirror is not available in Hardware menu and anywhere else in fact.
There is a mistake somewhere than. 

**General information**
 - OpenTX version 2.3.11
 - Hardware Frsky X9D+ 2019

3djc, 2021-04-16:
The documentation you are refering to is for X9D and X9D+, not X9D+ 2019. The 2019 does not offer this

closed 2021-04-16 by 3djc

RMFPV, 2021-04-16:
There is no other documentation. The difference is not described anywhere.

3djc, 2021-04-16:
I edited the doc before replaying to you, please refresh this page

RMFPV, 2021-04-16:
Great! Thanks!

Does it mean no external BT support for Taranis 19?
That's killing.