opentx/opentx #8455: Questions about TRxx sources in the Mixer when the trainer signal is lost

LapinFou, 2021-04-18:
Hi folks,

One member of the RCGroups forum discovered this following behavior.

The context: Two OpenTX radio are connected through a trainer cord plug on DSC ports.

On the **student** radio, the mixe 8 is defined as follow:
`CH08 = MAX (Ratio +100%).`

On the **master** radio, the prog is as follow:
`CH6 = TR8 (Ratio +100%)`
`L01 = |CH6| > 10%`
`L01 = |TR8| > 10%`
When the trainer cord is present **L01** & **L02** are **True**. 👍

In case of trainer signal lost (broken trainer connection), **L01** becomes **False** however **L02** stays **True**! 😳
Is it the intend behavior?
Does it mean in case of DSC signal lost all TRxx sources are forced to 0%.
If the answer is yes, can we rely on this behavior for the future version of OpenTX?
If the answer is no, then I guess it is a bug?

Seb

kilrah, 2021-04-18:
TRx lines get disabled when there's no signal, like if the line had a a switch set and that was off.
Use the usual "line logic" to define what happens in that case.

LapinFou, 2021-04-18:
If I got it correctly, it means the TRx source is a kind of NaN (or similar).
In the mixe there is an implicit 0% at the beginning.
In others words, I will get 0% on the mixe and get the latest known value on the inputs and logical switches. Did I got correctly?

kilrah, 2021-04-18:
> In others words, I will get 0% on the mixe and get the latest known value on the inputs and logical switches.

Correct

closed 2021-04-18 by kilrah

Nihonski, 2021-05-19:
I opened a similar issue just before.
My problem is, I need to have signal/value, which inducates, whether the signal is lost or not.
I use an X8R with SBUS to make a wireless trainer connection.
When the wireless connection is lost, all TR values are hold to the latest value, although the X8R sends failsafe data on the SBUS.
Is there anyway to find out, whether the TR values are up to date, or hold because of loss of signal?