opentx/opentx #8465: Crossfire Configure menu missing on 2.3.11-otx(4328055e)

c0debreaker, 2021-04-23:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Not sure what happened. I do remember seeing `Crossfire configure` menu. I'm only seeing this now. Did I miss a step when I upgraded to this version?

**Screenshots**
[image]
[image]

**General information**
 - OpenTX version 2.3.11-otx(4328055e
 - Jumper T16

3djc, 2021-04-23:
Is your external module set to CRSF ?

c0debreaker, 2021-04-23:
Yes. I am able to fly without any issue. I just want to add LQ into my DJI goggles OSD so I need to get inside nano configuration but I can't see it.

3djc, 2021-04-23:
Then you SD content is wrong

closed 2021-04-23 by 3djc

c0debreaker, 2021-04-23:
oh, maybe I am missing a step. Where can I find it?

c0debreaker, 2021-04-23:
I found it

c0debreaker, 2021-04-23:
I think 25 to 34 must be deleted so that it won't confuse users and all we need to do is download the latest sd card
```
Index of /2.3/release/sdcard/opentx-t16/
../
opentx-t18                                         06-Oct-2020 13:55                  10
sdcard-480x272-2.3V0025.zip                        12-Mar-2020 18:37           122580878
sdcard-480x272-2.3V0026.zip                        27-Mar-2020 20:23           121869033
sdcard-480x272-2.3V0029.zip                        13-Jun-2020 13:43           123215475
sdcard-480x272-2.3V0034.zip                        06-Oct-2020 13:39           124739825
sdcard-480x272-2.3V0035.zip                        08-Jan-2021 14:27           121241644
```

3djc, 2021-04-23:
if you pick the wrong one, the radio will tell you....

c0debreaker, 2021-04-23:
oh got it

c0debreaker, 2021-04-23:
it's working now. I just copied all the files under TOOLS directory. Thanks a lot!