opentx/opentx #8466: Unexpected Termination of Companion on Converting an OTX File

Hartinger, 2021-04-23:
I am using Companion 2.3.11 with several Radio profiles, i.e. Taranis X9D+ 2019, Taranis X7S ACCESS and Horus X10S Express. As I have a number of models that I fly with different Radios, I do regular conversions of the otx files for comparison, harmonisation, etc. In this context, I noticed the following erroneous behaviour.

If I work in the profile for the Taranis X7S ACCESS, and then open a Taranis X9D+ 2019 otx file and confirm the file conversion, Companion terminates withour any pre-warning and without saving anything. This problem does not occur if I open a Horus X10S Express otx file.
I have generated a test otx file in the Taranis X9D+ 2019 profile, with only one simple model in it, and the problem still occurs. So it does not seem to be related to any specific model setup.

elecpower, 2021-04-23:
On Win 10, I have tested on my model file converting from TX16S to X9D+ 2019 to X7S Access and no crash.
Also tested a X9D+ 2019 new file and wizard created model then converted to X7S Access and no crash.

@Hartinger what operating system and version are you using? Can you upload your simple X9D+ 2019 otx file so I can try and reproduce.

Hartinger, 2021-04-23:
I am using Win 10 pro, here is the simple otx file (you need to remove the .txt suffix).
[Test.otx.txt](https://github.com/opentx/opentx/files/6367782/Test.otx.txt)

elecpower, 2021-04-23:
@Hartinger thanks I can reproduce with your test file

elecpower, 2021-04-23:
@Hartinger  when I load your test file the conversion message says **X9D+** not **X9D+ 2019** and I have confirmed that the message is different. Here is the message when opening your test file. Can you please confirm what format the test file is in.

[image]

elecpower, 2021-04-23:
@Hartinger I have updated my previous post. I misread the msg. They are the same no matter whether X9D+ or X9D+ 2019.

elecpower, 2021-04-23:
Cause: The conversion process does not know the radio variants for all the sub types. In this case the 2019 has one extra switch and if it is used in a model config the conversion crashes.

Hartinger, 2021-04-24:
So we were very lucky that the simple model I used to test includes the additional button SI. :-)
Good to hear you could find the cause, thanks a lot.

PR #8469 (2021-04-24): Stop Companion crash when conversion switch index out of range
Fixes #8466 
Fixes #8403
Conversion unable to identifying the correct sub board type from FourCC that have differing hardware capabilities. eg X9D+ 2019 has one more switch than other X9D boards.
closed 2021-04-24 by 3djc