opentx/opentx #8467: Missing repetition option for global/special function "play music" on RadioMaster TX12 - opentx 2.3.11

GoHardt, 2021-04-23:
There is no option of choosing the amount of repetitions of the special or global function "play music" on the Radiomaster TX12. This leads to the situation that the choosen sound keeps playing on repeat if the choosen switch is active.

Steps to reproduce the behavior:
1. Click "Sys" button
2. Scroll with 'page' to Page 4/7 "global functions"
3. Create function "PlayTrack" for any button and any sound
4. See error of missing repetition option.

A clear and concise description of what you expected to happen.:

There should be the counter of repetitions as shown in screenshot 1 for the option play sound (=german: Spiel Töne)

**Screenshots**
Spiel Töne = Play Sound
Spiel Musik = Play Track

[image]

**General information**
 - OpenTX version 2.3.11
 - Hardware: Radiomaster TX12
 - OpenTX Companion 2.3.11

3djc, 2021-04-23:
Isn't that by definition the difference between play track and play music ?

GoHardt, 2021-04-23:
The german "Start Musik" is the equivilant for the english play track. There is no option as play music according to the manual

[image]

Page 45-46
https://manuals.plus/wp-content/plugins/pdfjs-viewer-shortcode/pdfjs/web/viewer.php?file=/wp-content/sideloads/radiomaster-tx12-manual-optimized.pdf&dButton=true&pButton=true&oButton=false&sButton=true#zoom=auto&pagemode=none

3djc, 2021-04-23:
ok, then it is likely a translation issue, because play track does have it

[image]

GoHardt, 2021-04-23:
ok well. So i guess i need to swap to the english version. i also figured i can apply that option in companion. thanks :)

closed 2021-04-23 by GoHardt