opentx/opentx #8473: Companion Write Models & Settings to Horus X10S Express

lazyfour, 2021-04-26:
When using Companion to Write Models & Settings to a Horus X10S Express radio the expected dialogue box:

[image]

does not appear and so the user does not have the ability to:
Check Firmware Compatibility
Backup Before Write
Patch HW Settings From Profile

The model is written directly to the radio successfully but with no dialogue options box or the subsequent green bars showing the progress of the write action which you get with other radios.

3djc, 2021-04-26:
All of those are only applicable to radio with EEPROM, so by design do not apply to colorlcd radios

elecpower, 2021-04-26:
The status bar advises if the writes were successful though it does get cleared after a few seconds. Maybe the consistent use of a progress bar could be considered for an enhancement.

3djc, 2021-04-26:
Yeah, we just need to keep in mind that the general principle, no message means no error, but yeah, the write successfull should stay until next user interaction idealy :)

lazyfour, 2021-04-26:
Thanks for reply guys. I was unaware of the difference between eprom and colour lcd radios, is this documented anywhere? Wouldn’t it be sensible to standardise the options across all radios? The user probably doesn’t care whether the data is stored on eprom or SD card. There is no visible status bar when I write to the Horus. Perhaps with a fast computer it happens so quickly as to be invisible? I would also like to be able to save a backup before writing a model and settings to the Horus. Is there no way to automate this in Companion?

Regards,

Mike

lazyfour, 2021-04-26:
I forgot to say that it is also very confusing that when you change radio profiles, the application menu remains unchanged so the locations for SD card and Flash screen are for the wrong radio. I would expect changing profile to also change the corresponding application and simulator menus.

kilrah, 2021-04-26:
SD and splash are on the radio page and do change with the profile.

lazyfour, 2021-04-26:
In Companion Settings, the Automatic Backup Folder and User Splash Screens folders in Application Settings do not change when the Radio Profile is changed. If I am working with a Taranis transmitter profile I would not expect to see references to Horus files and vice versa. I admit I am new to OpenTX so perhaps I don’t understand the design logic. I think OpenTX is wonderful but the documentation often is not or at least lags far behind the current software release.

elecpower, 2021-04-26:
Each profile can have a different SD Structure path

3djc, 2021-04-27:
Users absolutely needs to care and understand where data is located on radio. If you have a colorlcd radio, then you need to organise some kind of sdcard backup strategy

lazyfour, 2021-04-27:
I disagree. The user interface design should be concerned with the data 
and not the technology used to store it. Mobile phone users can have 
their texts, emails and photographs safely backed-up automatically by 
the application program (in IOS or Android) and they don't need to know 
whether it is being stored on eprom, SD card or some other cloud storage 
media so long as it is recoverable and the integrity is preserved. 
Ideally an OpenTX user would not have to know which options were 
applicable to their radio as these options would be changed for them 
simply by reference to the radio make and model type and they would then 
ONLY see applicable options. Most of the issues I have raised only occur 
because I am managing two different radio transmitters with the same 
copy of OpenTX Companion. Provided that there is clear and accurate 
documentation to explain the available options then I can manage the way 
I use them and I am happy to leave it there and for you to close my 
issue on Github. Thank you for your time and support.

Regards,

Mike

X10S Express (#8473)

<https://github.com/opentx/opentx/issues/8473#issuecomment-827346559> ,

closed 2022-04-16 by stale[bot]