opentx/opentx #8477: "Wrong signature" when flashing ELRS from X12S

trigger7, 2021-04-29:
When trying to flash ExpressLRS on a R9M module from a Horus X12S, I get a "wrong signature" error.

The .frk bootloader flashes fine.
The error happens during the next step, flashing the .elrs firmware using the "Flash External ELRS" action.

Maybe linked to #7221 ?

 - OpenTX version 2.3.12N487
 - Frsky X12S (with inverter mod and internal ISRM)
 - R9M ACCST (with resistor mod)
 - ELRS 1.0.0 RC2

trigger7, 2021-04-29:
It's the first time I tried to flash ELRS, so it never worked.

I can flash the .frk file (back and forth between the Frsky one and the ELRS one), but the second file (.elrs) always give me this error.

I tried both with the old internal ACCST module, and the newer ACCESS one (no idea if it makes a difference).

I do not own any other radio capable of flashing ELRS, so I can't compare (just an old 9XR not supported anymore after openTX 2.1).

tientrung113, 2021-05-02:
Hi. I am also getting the same signature error on my x12s. I used qx7 and loaded it. But the deadlock on rx r9mm two lights are on after loading start program.

trigger7, 2021-05-02:
Just to a message to say I'm ready to help in any way I can to get to the bottom of this issue. I am ready to build and flash modified firmwares if necessary.

I had a quick look at radio/src/io/multi_firmware_update.cpp
It seems the x12s is getting past the first few checks (otherwise we would get "NoSync" or "NoSignature"). It seems able to read some form of signature, just not the expected one.
Would it be helpful, for a start, to find you exactly what is the x12s reading from the module ?

carlic63, 2021-05-02:
I have the same issue with my X12S and R9M 2019.
At first I got "No sync" error, now I only get "wrong signature".

trigger7, 2021-05-02:
As a quick test, I tried flashing the ELRS bootloader (r9m_elrs_bl.frk) with an airlink usb dongle, then got the same result when trying to flash the firmware.elrs on the radio.
Which to me, seems to confirm the issue is on the multi_firmware_update side of things (and not on the frsky_firmware_update).
I'll try to compile a custom opentx to show exactly what signature is being read from the module.

tientrung113, 2021-05-02:
Yes. Thanks for the wisdom you have. I went 35km to borrow a qx7 tx and got it back an hour later. the module problem has been solved. but there is a problem with rx r9mm. After loading the file via tx, the 2 status lights are on. Press the binding button, the 2 lights flash in turn. Above the guide has given me a chat rom something but I don't know where to get the fix file. help me...

carlic63, 2021-05-03:
Did anyone try the STlink route?

raphaelcoeffic, 2021-05-03:
Ok, I'll try to dig out a X12S somewhere, but It might take time until I can make it (next time I see my buddy with his X12S).

carlic63, 2021-05-03:
@trigger7 why do you use inverter mod? I thought you only need it to unlock the 500Hz in the 2.4GHz band.
900MHz is only good up to 200Hz anyway?

raphaelcoeffic, 2021-05-03:
In case I see my friend with the X12S, is there anything I would have to do on a stock X12S to test that?

carlic63, 2021-05-03:
Not sure why this is neccessary for flashing but according to this link we need this inverter mod for the transmitter.
https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/#comment-8763

Also probably a custom OpenTx build beforehand? Not sure....

trigger7, 2021-05-03:
I managed to build opentx with a quick and dirty hack to display the signature read from the module.
Before flashing, I have a couple of questions. I asked on the discord channel, but didn't get an answer yet.

I had done the inverter mod a while ago, because I was considering getting a crossfire module at the time (and the inverter itself is dirt cheap). Not sure if it's needed for ELRS, but a faster inverter shouldn't hurt even at lower baudrates.

tientrung113, 2021-05-03:
Trigger7# sen to mail opentx hack x12s. Tientrung11352@gmail.com.    thank bro

trigger7, 2021-05-03:
Managed to run my modified firmware.

Weirdly, the 4 bytes signature received change each reboot...
So far, I got :
`0x10 0x14 0x03 0x97`
then :
`0x10 0x14 0x1E 0x97`
(then the first one again, then the second one again)

0x10 seems to correspond to STK_OK and 0x14 to STK_INSYNC (from radio/src/io/stk500.h). Or is it a coincidence ?

trigger7, 2021-05-03:
So, I hacked a little bit more :

After adding a small wait between requesting and reading the signature, I always seem to get the same answer every time (the one with 0x1E).

Now, if I discard the 0x10 0x14, the next 4 bytes look like a valid signature :
`0x1E 0x97 0x02 0x10`

I repeated the process a few times, always got this signature.

I can't explain where do the extra 2 bytes ( STK_OK STK_INSYNC ) come from, so I guess we would first need to clear that up before implementing a proper fix...

My next step will probably be to actually flash ELRS with my modified firmware and see if it works

tientrung113, 2021-05-03:
great. But what you interfered with the firmware and I don't know anything. looking forward to a complete fix. or is there a document you can lead me there.

trigger7, 2021-05-03:
Bad news, I wasn't able to flash the ELRS firmware on the R9M.

Even after passing the signature check (with my modification), I get a "NoPageSync" error while it tries to write the firmware.

I'm now over my head and would need someone who actually knows what they're doing to take over...

tientrung113, 2021-05-03:
[image] qx7.

trigger7, 2021-05-03:
Good news everyone, I manage to flash my R9M module with the X12S !

From what I understand, the X12S has DMA capabilities that other radios haven't.
When openTX initialize a new serial/telemetry port, DMA is activated by default on the X12S.
By not activating DMA on the port used to flash the external module, I was able to successfully write the ELRS firmware.

Still no idea why DMA has a different behaviour however...

I'll talk to the openTX devs on discord and prepare a pull request.

tientrung113, 2021-05-03:
can send me firmware file to email (tientrung11352@gmail.com). I stay up all night waiting for this good.

Dannyc99, 2021-05-03:
Awesome job, any chance you can host the firmware somewhere for us to grab

raphaelcoeffic, 2021-05-04:
@Dannyc99 once merged, this will be the nightlies.

tientrung113, 2021-05-04:
Has it been pulled yet? I just tried to see a 2am revision on May 4th still signed ...
I just joined so I don't know much.

Dannyc99, 2021-05-04:
I downloaded the nightly that just came out today but I don't think that is
in there yet and of course I forgot to make a backup of my SD card LOL any
clues as to when this will be in the nightly or I guess out of the nightly
LOL

trigger7, 2021-05-05:
The openTX devs are looking into the issue, as they think the problem might run deeper, and my fix just hides it.

I know this issue has been bugging me for a while and preventing me from using my hardware properly, so I'm tempted to share the patched firmware...

I would strongly suggest waiting for the opentx devs to have a proper look and merge a fix (mine or a better one) into the nightlies.

If you want to test my fix, I would encourage you to download the source code from my branch, review my modifications and compile your own firmware.

If you insist on installing an untested firmware from an unknown origin, knowing that neither me nor the opentx team can be held responsible if anything bad happens to your radio, here is the link :
https://drive.google.com/drive/folders/1rIO_k3PQPL8LFGdvr4ESbXQlW9c45yFw?usp=sharing
There is a regular version for internal accst module, and one for internal access (although it should not make any difference if you just want to flash your external module).
You may have to turn the external module off then on after flashing (or flash while the module is off).
You also should revert back to your previous firmware after flashing your external module (as the fix has no influence while using the module).
**This firmware only fixes the "Wrong signature" error while trying to flash an external module (ExpressLRS and possibly MPM). If your issue is different, it will not fix it and you should not use this firmware.**

If you think sharing the patched firmware is wrong, against usual practices or otherwise, let me know and I'll remove the link.

tientrung113, 2021-05-05:
If I am with you now I will hug you dearly.
just to load the r9m module will return to 3.10, which lrs recommends.
and the latest 3.12 nightly version now I see x12s go into safe mode when the ppm protocol is set.

tientrung113, 2021-05-05:
> Nếu bây giờ anh ở bên em, anh sẽ ôm em thật yêu.
> chỉ để tải mô-đun r9m sẽ trở về 3.10, theo lời khuyên của cô.
> và phiên bản hàng đêm 3.12 mới nhất bây giờ tôi thấy x12s chuyển sang chế độ an toàn khi giao thức ppm được thiết lập.

sorry. Google translates my language too nonsense.

tientrung113, 2021-05-05:
I was successful. thank you again. I'm glad they worked after a long time researching this project with a brick lite r9m module

Dannyc99, 2021-05-07:
Its not a super secret firmware that makes our drones fly directly to your
house is it?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

carlic63, 2021-05-08:
I was wondering would the STlink tool to flash the firmware be a workaround for now?

trigger7, 2021-05-08:
You can use an stlink, but it's a one way ticket to ELRS. No possibility to flash back to FrSky if you ever need/want to.

carlic63, 2021-05-08:
Thats what I was afraid of.
So no fix for now?

trigger7, 2021-05-08:
You can pull my branch and compile a temporary opentx firmware with my fix/workaround (there are guides on the wiki for linux and windows. I used linux, and if you're familiar with it, it's actually fairly easy).

You can use the binaries I provided if you feel like it (despite my joke, I didn't do anything malicious).

Otherwise yes, you'll have to wait for the opentx team to investigate the issue and merge a fix into the opentx codebase.

tientrung113, 2021-05-08:
I'm working on a combo set of air units, rx r9mm plugs to link on all my 3 to 7 inch quads. Have you ever done that?

commit af17efad (2021-05-15): Fix #8477
closed 2021-05-15 by 3djc