opentx/opentx #8490: USB Serial Telemetry Mode

vimalb, 2021-05-08:
OpenTX has an existing feature to mirror telemetry output form the radio to an aux serial port. 

For radios which do not have a physically accessible aux serial port, I propose adding in a new USB connection mode "USB Telemetry" which mirrors telemetry output over the USB port functioning as a USB serial device.

This would be a particularly valuable feature which would let a broader range of radios connect to ground station computers / phones / tablets.

I have implemented a proof-of-concept of this feature here and tested it against a Radiomaster TX12: https://github.com/vimalb/opentx/commit/4d2b32fd772aeaa04939aeef33c9f01ea249e7e2

Specifically with these changes, I was able to connect my TX12 with a Crossfire module to an Android phone running [Telemetry Viewer](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry) and have telemetry data stream from my aircraft to my phone pretty seamlessly. 

I'm happy to convert the proof-of-concept to an actual PR implementing this feature per the OpenTX git workflow, but first wanted to make sure there was interest in this feature and also wanted some feedback on the POC changes since I'm new to working with OpenTX.

brainbubblersbest, 2021-05-09:
Olliw42 has implemented this already in his mav-link integration branch, as Part of it's mav-link Router Implementation.
This Enables Routing between the two aux_serial and the usb as Serial. Maybe that Feature can already be merged partially?

vimalb, 2021-05-09:
@brainbubblersbest - is https://github.com/opentx/opentx/pull/7702/files the branch/PR you are referring to? 

I was skimming though the diff and didn't see anything that looked obviously like supporting usb serial as a first-class aux_serial output, though maybe I am missing something?

If you could point me towards which exact branch/PR has this kind of functionality implemented, I'm more than happy to test it out.

brainbubblersbest, 2021-05-09:
Check the last ~50 posts

vimalb, 2021-05-09:
@brainbubblersbest - thank you for the pointers - after reading through both the thread and the source of https://github.com/olliw42/opentx/tree/mavtelem-dev I believe my proposed feature is slightly different from USB_MAVLINK_MODE 

Specifically USB_MAVLINK_MODE also uses usb_serial, but establishes a bidirectional mavlink connection and integrates with the rest of the mavlink implementation. 

My proposed USB_TELEMETRY mode is designed to be a lower-level unidirectional serial telemetry output that works for any of the other telemetry protocols as well, eg: frsky or crsf. It is literally the equivalent of UART_MODE_TELEMETRY_MIRROR but outputting over USB instead of an aux serial port.

Perhaps it's better to rename my proposed mode USB_MODE_TELEMETRY_MIRROR as that is how I would expect it to function.

brainbubblersbest, 2021-05-09:
Yes youre right. 
So i gave a +1 to this issue here.

3djc, 2021-08-02:
Not sure why I missed this one, love it

vyalyh-oleg, 2022-01-28:
+1 for this feature too.
It would be great to have ability to transfer over external UART(AUX) full-fledged MAVLink (I prefer to use Bluetooth over it). And also the configurable baud rate would be nice addition too.