opentx/opentx #8492: Request default model throttle curve to be linear (0~100% across stick range) instead of 50%

chronowarper, 2021-05-09:
This is totally user error, but we had a situation today where the quad was accidently left armed and the pilot was changing throttle profile in the radio. This flipped past a default profile where throttle was 50% across the stick travel, and the quad took off.

We're not sure why the default curves have '50%' across the throttle range. The safer option would be to have it linear (0~100%) across the stick travel.

raphaelcoeffic, 2021-05-09:
Something similar happened to me while trying to setups a gas curve while the model was turned on.

lshems, 2021-05-10:
Fully agree.

Miami-Mike, 2021-05-20:
That makes sense. What could be more benign than a curve with an output that echos the input? That is, f(x) = x.

Miami-Mike, 2021-11-16:
Since this went stale, I passed it on the [the EdgeTX forum](https://github.com/EdgeTX/edgetx/issues/1067).