opentx/opentx #8495: 2.4ghz ELRS Support for Radiomaster TX16 Internal multi module

evildronepilot, 2021-05-10:
Is it possible to get support for the internal multi-protocol module built into TX16.. radio?
Having the option to select the 2.4 ELRS and bind with, for example, ghost RX... would be a mind-blowing/game-changing feature even if we only have a low power lever available.

3djc, 2021-05-10:
Not sure what you are exactly referring to, but ghost rx use a modulation type not available in any of the 4  MPM chip

davifelix, 2023-05-26:
Good morning.  resuming this discussion...... would it be possible to have an internal 2.4ghz elrs module instead of the 4in1 module?  in order to have the free bay for the 900mhz module.

rotorman, 2023-05-27:
RadioMaster TX16S mkII is being sold with internal Express LRS option.

davifelix, 2023-05-27:
yes I know...... What I would like to know is if it is possible to modify the tx16 with the elrs module of MKII....... Could it work? Or are they just different?

rotorman, 2023-05-27:
Cannot follow unfortunately what you are asking - please try to rephrase. And also for such questions, best turn to OpenTX or EdgeTX Discord, as Github here is more for software/firmware developmental issues.

davifelix, 2023-05-27:
scrivo in italiano così evito due traduzioni. vorrei sapere se la scheda lrs 2.4 della tx16s MARKII può essere montata su una tx16s normale, eliminando la scheda 4in1. grazie

rotorman, 2023-05-28:
What you are asking is if TX16S mkII internal ExpressLRS module can be used in 1st generation TX16S - the answer is yes, it can. The mounting and the RF module connector of the internal RF module are the same between 1st gen. and mkII versions of TX16S.

davifelix, 2023-05-28:
esatto!! è quello che volevo sapere. grazie mille! non devo modificare il firmware della radio giusto? o devo caricare una versione per mkII?

rotorman, 2023-05-29:
If your 1st gen. TX16S is running a firmware that is new enough that it supports internal Express LRS, then no update is required, otherwise you need to update in order to use internal ExpressLRS.
In general, there might be other reasons to update as well, thus my recommendation would be to update your TX16S to the latest release version anyhow. For all further questions, please turn to OpenTX or EdgeTX Discord.