opentx/opentx #8497: Open Tx2.1.9  et Next step RC

PAT14440, 2021-05-11:
Bonjour,

Je suis constructeur de ma radio Next stepRC , elle fonctionne parfaitement et répond à mes attentes en fonctionnement avec Open TX 2.1.9, cependant un problème se pose, j'ai besoin d'un ralentissement de 10 seconde ssur une commade (ouverture et fermeture de trappes) et il est impossible d'aller au dessus de 7.5 secondes dans ce réglage. Y a t-il une raison amenée par cette version de Companion ou y a t-il moyen de changer ce paramètre dans le code source et par quel moyen? 
Avec Next stp rc je ne peux pas utiliser les dernières versions de OpenTX  car elle ne prennent pas en charge ma radio NextStep RC. 
Merci d'avance pour vos réponses

A+

Miami-Mike, 2021-05-20:
"I am the manufacturer of my Next stepRC radio, it works perfectly and meets my expectations in operation with Open TX 2.1.9, however a problem arises, I need a 10 second slowing down on a command (opening and closing hatches) and it is impossible to go above 7.5 seconds in this setting. Is there a reason brought by this version of Companion or is there a way to change this setting in the source code and by what means?
With Next stp rc I cannot use the latest versions of OpenTX because it does not support my NextStep RC radio.
Thank you in advance for your answers"

I don't see any problem with the OpenTX 2.1.9 Slow function. I set up an input that toggles between -100% and +100%, and a mix with Slow Up and Slow Down set to 10 seconds, and I timed it with a stopwatch. It takes 10 seconds to travel from one end to the other.

PAT14440, 2021-05-20:
Bonjour,
Merci beaucoup pour votre réponse, mais mas version de Open TX 2.1.9 ne peut aller au dela de 7.5secondes , ou puisje trouver votre version . 
Merci d'avance 
[image]

Miami-Mike, 2021-05-20:
Which radio? / Quelle radio?

[image]

PAT14440, 2021-05-20:
[image: image.png]

Thank in advance ,, can you give me your version Of Companion 2.1.9

As Soon
Pascal

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Miami-Mike, 2021-05-20:
My version of OpenTX 2.1.9 is available at https://www.open-tx.org/2016/09/15/opentx-2.1.9

PAT14440, 2021-05-20:
[image: image.png]

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kilrah, 2021-05-20:
He's on the "DIY Mega2560".
AVR radios used 5 bits for slow/delay, so -16 to +15 *0.5s, limited to +/-7.5 for consistency.

You can maybe chain multiple mixesto achieve more slow.

closed 2021-05-20 by kilrah