opentx/opentx #8504: "Trio Control"  RSSI and VFR transmitted from multiple receivers at the same time with telemetry

Devil-67, 2021-05-13:
Good morning on Father's Day ;-)

A question, could the DEVS at ACCESS radios with "Trio Control" make it possible to transmit RSSI and VFR from all connected receivers via telemetry?
It will probably be possible because each receiver sends RSSI and VFR with a different ID in redundancy mode.
In my case RX1 RSSI with ID24 and VFR with ID26, RX2 with ID 56 and VFR with ID 58 and RX 3 RSSI with ID 88.
That would be nice to find good antenna positions and to easily localize errors.
Problems with RX1, 2 or 3 can also be easily identified in the later evaluation in the LOG.

I am curious about your answer 

Have a nice father day

3djc, 2021-05-13:
Thats what FrSky said they would do, and never did so far. Really nothing we can do, it is RF protocol related, nothing to do with OTX

closed 2021-05-13 by 3djc

Devil-67, 2021-05-13:
Ok I understand.
thanks