opentx/opentx #8505: Taranis X-Lite Key Remapping

K1rdro, 2021-05-13:
The navigation joystick on the various models of the Taranis X-Lite radios is a point of failure. This joystick is used to navigate through the various menus, and is the Enter key when it is pressed. When the joystick stops working consistently, navigating through the menus and making any selections begins to range between frustrating and impossible.

A member of the RC Groups forum made a custom firmware version where the joystick and directional pad functions were swapped. The enter key was then swapped with the shift key. I'd like to see the option to use this configuration in mainline opentx, since the custom firmware was not made for each version of X-Lite. 

The current solution from Frsky is to have the controller sent in for costly repairs (where they replace the main board with one that will fail just as quickly), or to DIY the main board replacement.

[Simon Ernst](https://simonernst.com/) is the RC Groups user 527FPV, who has the custom firmware already written and available at [His site](https://downloads.simonernst.com/xlite/). He sounds like he'd be interested in supporting the change. Here's a link for more context: [RCGroups Forum Discussion](https://www.rcgroups.com/forums/showthread.php?3080337-FrSky-New-Taranis-X-Lite-S-and-Pro-X9-Lite-S-R9M-Lite-and-Pro-MPM-Lite/page105)

bigtwin, 2021-05-14:
How about GDPR?

K1rdro, 2021-05-14:
You mean this GDPR: https://gdpr-info.eu/?  If not, then ignore this. If so, no data is being processed. I'm just referencing information openly provided by the individual, and when their website is in their own name, it's kinda hard to expect anonymity.

bigtwin, 2021-05-15:
Yes, I was referring to this and thx for responding! I am not familar with all circumstances about Simon and 527FPV. Its just the first time I noticed that a thrid party is clearly linking a real name to a nickname, which is forbidden in every forum I am reading and surley critical in terms of GDPR. 
However, if you are aware of this and all is fine, the purpose of my hint is fullfilled.

Knifa, 2021-05-22:
I just did this myself! It's an easy change to `radio/src/targets/taranis/hal.h`.
I'd love for it to be a standard option, though, sure.

```patch
diff --git radio/src/targets/taranis/hal.h radio/src/targets/taranis/hal.h
index f70e1e690..cc21f8408 100644
--- radio/src/targets/taranis/hal.h
+++ radio/src/targets/taranis/hal.h
@@ -32,21 +32,26 @@
   #define KEYS_GPIO_REG_ENTER           GPIOF->IDR
   #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_0  // PF.00
 #elif defined(PCBXLITE)
+  // Joystick click (Enter) -> shift
+  // Exit -> Enter
+  // Shift -> Exit
   #define KEYS_GPIO_REG_SHIFT           GPIOE->IDR
-  #define KEYS_GPIO_PIN_SHIFT           GPIO_Pin_8  // PE.08
+  #define KEYS_GPIO_PIN_SHIFT           GPIO_Pin_11 // PE.11
   #define KEYS_GPIO_REG_EXIT            GPIOE->IDR
   #define KEYS_GPIO_PIN_EXIT            GPIO_Pin_7  // PE.07
   #define KEYS_GPIO_REG_ENTER           GPIOE->IDR
-  #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_11 // PE.11
-  #define KEYS_GPIO_REG_UP              GPIOE->IDR
-  #define KEYS_GPIO_PIN_UP              GPIO_Pin_10 // PE.10
-  #define KEYS_GPIO_REG_DOWN            GPIOE->IDR
-  #define KEYS_GPIO_PIN_DOWN            GPIO_Pin_14 // PE.14
-  #define KEYS_GPIO_REG_LEFT            GPIOE->IDR
-  #define KEYS_GPIO_PIN_LEFT            GPIO_Pin_12 // PE.12
-  #define KEYS_GPIO_REG_RIGHT           GPIOE->IDR
-  #define KEYS_GPIO_PIN_RIGHT           GPIO_Pin_13 // PE.13
-#elif defined(RADIO_T12)
+  #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_8  // PE.08
+
+  // Use DPAD
+  #define KEYS_GPIO_REG_UP               GPIOB->IDR
+  #define KEYS_GPIO_PIN_UP               GPIO_Pin_0  // PB.00
+  #define KEYS_GPIO_REG_DOWN             GPIOB->IDR
+  #define KEYS_GPIO_PIN_DOWN             GPIO_Pin_1  // PB.01
+  #define KEYS_GPIO_REG_LEFT             GPIOC->IDR
+  #define KEYS_GPIO_PIN_LEFT             GPIO_Pin_4  // PC.04
+  #define KEYS_GPIO_REG_RIGHT            GPIOC->IDR
+  #define KEYS_GPIO_PIN_RIGHT            GPIO_Pin_5  // PC.05
+#elif defined(RADIO_FAMILY_JUMPER_T12)
   #define KEYS_GPIO_REG_EXIT            GPIOD->IDR
   #define KEYS_GPIO_PIN_EXIT            GPIO_Pin_2  // PD.02
   #define KEYS_GPIO_REG_ENTER           GPIOE->IDR
@@ -216,14 +221,15 @@
   #define TRIMS_GPIO_REG_RHR            GPIOC->IDR
   #define TRIMS_GPIO_PIN_RHR            GPIO_Pin_13 // PC.13
 #elif defined(PCBXLITE)
-  #define TRIMS_GPIO_REG_LHL            GPIOC->IDR
-  #define TRIMS_GPIO_PIN_LHL            GPIO_Pin_4  // PC.04
-  #define TRIMS_GPIO_REG_LHR            GPIOC->IDR
-  #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_5  // PC.05
-  #define TRIMS_GPIO_REG_LVU            GPIOB->IDR
-  #define TRIMS_GPIO_PIN_LVU            GPIO_Pin_0  // PB.00
-  #define TRIMS_GPIO_REG_LVD            GPIOB->IDR
-  #define TRIMS_GPIO_PIN_LVD            GPIO_Pin_1  // PB.01
+  // Use joystick
+  #define TRIMS_GPIO_REG_LHL            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LHL            GPIO_Pin_12 // PE.12
+  #define TRIMS_GPIO_REG_LHR            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_13 // PE.13
+  #define TRIMS_GPIO_REG_LVU            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LVU            GPIO_Pin_10 // PE.10
+  #define TRIMS_GPIO_REG_LVD            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LVD            GPIO_Pin_14 // PE.14
 #elif defined(RADIO_X7ACCESS)
   #define TRIMS_GPIO_REG_LHR            GPIOD->IDR
   #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_15 // PD.15
```

Knifa, 2021-05-22:
Here's a patch that'll do it, to get the ball rolling. I'm not sure what else is needed to make it a proper main-line option, unfortunately.

```patch
diff --git radio/src/targets/taranis/hal.h radio/src/targets/taranis/hal.h
index f70e1e690..e0cc9c26a 100644
--- radio/src/targets/taranis/hal.h
+++ radio/src/targets/taranis/hal.h
@@ -32,20 +32,24 @@
   #define KEYS_GPIO_REG_ENTER           GPIOF->IDR
   #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_0  // PF.00
 #elif defined(PCBXLITE)
+  // DPAD -> Navigation
+  // Joystick click -> Shift
+  // Right-most button -> Enter
+  // Left-most button -> Exit
   #define KEYS_GPIO_REG_SHIFT           GPIOE->IDR
-  #define KEYS_GPIO_PIN_SHIFT           GPIO_Pin_8  // PE.08
+  #define KEYS_GPIO_PIN_SHIFT           GPIO_Pin_11 // PE.11
   #define KEYS_GPIO_REG_EXIT            GPIOE->IDR
   #define KEYS_GPIO_PIN_EXIT            GPIO_Pin_7  // PE.07
   #define KEYS_GPIO_REG_ENTER           GPIOE->IDR
-  #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_11 // PE.11
-  #define KEYS_GPIO_REG_UP              GPIOE->IDR
-  #define KEYS_GPIO_PIN_UP              GPIO_Pin_10 // PE.10
-  #define KEYS_GPIO_REG_DOWN            GPIOE->IDR
-  #define KEYS_GPIO_PIN_DOWN            GPIO_Pin_14 // PE.14
-  #define KEYS_GPIO_REG_LEFT            GPIOE->IDR
-  #define KEYS_GPIO_PIN_LEFT            GPIO_Pin_12 // PE.12
-  #define KEYS_GPIO_REG_RIGHT           GPIOE->IDR
-  #define KEYS_GPIO_PIN_RIGHT           GPIO_Pin_13 // PE.13
+  #define KEYS_GPIO_PIN_ENTER           GPIO_Pin_8  // PE.08
+  #define KEYS_GPIO_REG_UP              GPIOB->IDR
+  #define KEYS_GPIO_PIN_UP              GPIO_Pin_0  // PB.00
+  #define KEYS_GPIO_REG_DOWN            GPIOB->IDR
+  #define KEYS_GPIO_PIN_DOWN            GPIO_Pin_1  // PB.01
+  #define KEYS_GPIO_REG_LEFT            GPIOC->IDR
+  #define KEYS_GPIO_PIN_LEFT            GPIO_Pin_4  // PC.04
+  #define KEYS_GPIO_REG_RIGHT           GPIOC->IDR
+  #define KEYS_GPIO_PIN_RIGHT           GPIO_Pin_5  // PC.05
 #elif defined(RADIO_T12)
   #define KEYS_GPIO_REG_EXIT            GPIOD->IDR
   #define KEYS_GPIO_PIN_EXIT            GPIO_Pin_2  // PD.02
@@ -216,14 +220,15 @@
   #define TRIMS_GPIO_REG_RHR            GPIOC->IDR
   #define TRIMS_GPIO_PIN_RHR            GPIO_Pin_13 // PC.13
 #elif defined(PCBXLITE)
-  #define TRIMS_GPIO_REG_LHL            GPIOC->IDR
-  #define TRIMS_GPIO_PIN_LHL            GPIO_Pin_4  // PC.04
-  #define TRIMS_GPIO_REG_LHR            GPIOC->IDR
-  #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_5  // PC.05
-  #define TRIMS_GPIO_REG_LVU            GPIOB->IDR
-  #define TRIMS_GPIO_PIN_LVU            GPIO_Pin_0  // PB.00
-  #define TRIMS_GPIO_REG_LVD            GPIOB->IDR
-  #define TRIMS_GPIO_PIN_LVD            GPIO_Pin_1  // PB.01
+  // Use joystick
+  #define TRIMS_GPIO_REG_LHL            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LHL            GPIO_Pin_12 // PE.12
+  #define TRIMS_GPIO_REG_LHR            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_13 // PE.13
+  #define TRIMS_GPIO_REG_LVU            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LVU            GPIO_Pin_10 // PE.10
+  #define TRIMS_GPIO_REG_LVD            GPIOE->IDR
+  #define TRIMS_GPIO_PIN_LVD            GPIO_Pin_14 // PE.14
 #elif defined(RADIO_X7ACCESS)
   #define TRIMS_GPIO_REG_LHR            GPIOD->IDR
   #define TRIMS_GPIO_PIN_LHR            GPIO_Pin_15 // PD.15
```

K1rdro, 2021-05-24:
> The joystick is a crime against humanity.

I lol'd at this. Exactly how I feel every time I have to change a parameter on my radio in the field.

Knifa, 2021-05-24:
> > The joystick is a crime against humanity.
> 
> I lol'd at this. Exactly how I feel every time I have to change a parameter on my radio in the field.

My very own joystick broke so I'd been using Simon's for a while before myself. Got another X-Lite and now I can't go back! Would love to see this as a standard option too.

erstec, 2021-05-30:
> Here's a patch that'll do it, to get the ball rolling. I'm not sure what else is needed to make it a proper main-line option, unfortunately. The joystick is a crime against humanity.
@Knifa Be sure to have button assignments as default was in bootloader. Or you can't enter it at all. It is really hard to press and "fold" joystick at same time )
At least, I am using standard bootloader with same as yours modified custom fw.

Knifa, 2021-05-30:
> > Here's a patch that'll do it, to get the ball rolling. I'm not sure what else is needed to make it a proper main-line option, unfortunately. The joystick is a crime against humanity.
> 
> @Knifa Be sure to have button assignments as default was in bootloader. Or you can't enter it at all.
> At least, I am using standard bootloader with same as yours modified custom fw.

Oh yeah, that's a good point. I'm just using the default bootloader still so no problems. Pretty dangerous to flash I guess!