opentx/opentx #8509: Horus X12S Emergency Mode with oversize model image & inav Lua

FLYN-001, 2021-05-14:
Crashes OpenTX! Emergency Mode! Changing theme & theme colours multiple times on a model with iNav Lua running & oversized model image

**To Reproduce**
Steps to reproduce the behavior:
1. Select slightly oversize (pixel) model image
2. install & inav Lua 1.7.4 on second widget screen, display first widget screen with trims sliders
3. change to different themes & change colour of flexi theme multiple times
4. OpenTx crashes putting tx into emergency mode - intermittent error

**Expected behavior**
not to crash OpenTX

**General information**
 - OpenTX version 2.3.11
 - Hardware FRSKY X12S ACCESS
 - iNav Lua 1.7.4
 

[image]

raphaelcoeffic, 2021-05-15:
opentx 2.3 is missing proper clipping. Most of the time, when you write off screen, you will crash the radio.

FLYN-001, 2021-05-15:
Ah ok that'll explain it no doubt, it seems to be running better with correctly sized images now, however the over pixel size images would still display correctly in the model image box on screen giving no indication they were the incorrect size.

raphaelcoeffic, 2021-05-16:
That is because the model widget resizes image to the correct size. Not sure the function is available in LUS.

closed 2022-04-16 by stale[bot]