opentx/opentx #8516: Qx7 Hanging And Freezing 2.3.11

JustSushmit, 2021-05-20:
Hey There I Just Updated My Qx7 to 2.3.11 and it sometimes freeze or hang..

Issue Description When I Power It Dots Appear And After All Dots Appear Nothing Happens and Suddenly after 1 -2 Second directly coming to Homescreen with no welcome Speech and Also other button also don't work or freeze

Changes - No hardware Change I updated my Internal Module XJT

3djc, 2021-05-20:
We have a VERY large number of x7 running 2.3.11 without issues. Please reflash your radio, check your sd, the issue is on your side.

closed 2021-05-20 by 3djc

JustSushmit, 2021-05-20:
This Issue happens when you reduce boot time if you keep boot time to default 2S it doesn't happen

JustSushmit, 2021-05-20:
https://youtu.be/J8Vqm73JRY4 look at This video I tried 2.3.10 it doesn't happen on it and u tried flashing 2.3.11 many times from companion

JustSushmit, 2021-05-20:
> We have a VERY large number of x7 running 2.3.11 without issues. Please reflash your radio, check your sd, the issue is on your side.

What do you think about the video

3djc, 2021-05-20:
I tried with 0s boot time, no issue, 0s + no splash, no issue. Please go to comunity sites like rcgroups or openrcforums for comunity support

JustSushmit, 2021-05-20:
I reflashed my tx with fresh companion installation did everything correctly updated boot loader formatted EPPROM did possibly everything did same thing on other qx7 same result it's hanging but splash screen and power on delay has nothing to do with it as on default also it happens but it Only happens if you switch on and off tx multiple times within a minute

bigtwin, 2021-05-20:
Multiple power-on and power-offs in a minute causes a hang on QX7 with 2.3.11, but not with 2.3.10. Did I get this correct?
If so, I might try to recreate. I have a QX7 ACCST and it never ever got hung or freeze. But actually I cannot think of a situation where I powered it on/of multiple times in a short period of time.

JustSushmit, 2021-05-21:
Yup

JustSushmit, 2021-05-21:
It happens once in like 20 random power cycles within short period I formatted EPPROM and did possibly everything now all buy some sd cards class 10 and try those actually my Power button was bit fiddly so I was testing it during which I discovered this

bigtwin, 2021-05-21:
Ok, I have powered on and off my QX7 several times in a row. Nothing happened, but it felt real bad. It simply hurt myself. Usually noone would try to power on and off a computer that often in such a short time period. However, my QX7 worked as expected and I dont see a reason why I should continue to switch it on and off like a flash light. This is definetly out of scope for such a device. Totally independend of the fact that this might was possible with 2.3.10.
Pardon me, but I am out of this re-creation.

JustSushmit, 2021-05-21:
Give me some time I'll replace sd card of both tx and see if This issue Happens even without short power cycle also I want to mention that This issue happened After I updated my XJT module of Qx7 to latest version of ACCST 1.0

JustSushmit, 2021-06-08:
I can confirm it happens randomly I can't recreate the Issue I changed sd card did multiple reflashes it's very random and it doesn't happen during flight it only happens when I boot if it's not there then entire flight session it won't happen but but if it happens only way is switching off and on and when it happens power dot freeze and then open tx logo comes and there's no welcome sound and then it starts to hang