opentx/opentx #8520: Taranis X-lite Pro Sliders as Throttle Bug, Sliders T-Reverse and Sliders as Throttle Timer Bug

AlphaCharlyX, 2021-05-21:
Hello together,

there is a problem with starting the Timer (Ths, TH% and THt) and the T-Reverse (Throttle-Reverse doesn't work) when you define S1 or S2 (Sliders) for you throttle-input in your Taranis X-Lite Pro. 

T-Reverse has no effect while using Throttle-Input with S1 or S2 sliders. The only way tho reverse Throttle is to set  in the Input-tab or in the Mixers-tab or Outputs-tab the weight of Throttle to -100. In this cases then the problem is, that the timers (Ths, TH% and THt) doesn't start although the throttle is on. The timers only start, if throttle is off. This should work the opposite way.

Another bug is on the mainscreen. The bars for S1 and S2 are working the wrong way. They should go up or down the opposite way. If you push S1 (left Slider) to the right the left Slider S1 bar should go up. If you push S2 (right Slider) to the left the right Slider S2 bar should go up.

Please fix this bug quickly.

Thx und Greetings Andreas

General information
 - OpenTX version (e.g. 2.3.11 or 4328055e)
 - Hardware Frsky Taranis X-lite Pro
 - OpenTX Companion 2.3.11 on Win10

kilrah, 2021-05-21:
For the timers you set your Throttle Source to your throttle channel and they will work as expected.
Bars are consistent with return values.

AlphaCharlyX, 2021-05-21:
> 
> 
> For the timers you set your Throttle Source to your throttle channel and they will work as expected.
> Bars are consistent with return values.

Have you tried it on a X-lite? On Campanion your solution works on the radio it doesn't.

3djc, 2021-05-21:
not throttle stick (or pots in your case), the channel used for throttle

AlphaCharlyX, 2021-05-21:
Yes I know,

but others have the same issue. [Check here](https://frsky-forum.de/index.php?thread/5366-x-lite-pro-slider-umkehren/)

AlphaCharlyX, 2021-05-21:
[image]
[image]
[image]

These are my settings on the X-lite Pro and on Companion. On Companion Timers work properly, on the radio they don't.

AlphaCharlyX, 2021-05-22:
> For the timers you set your Throttle Source to your throttle channel and they will work as expected.

If you do so the timers work as exspected, but then the Throttle-Warning doesn‘t work properly (This problem occurs on my Tarranis X10s Express as well)! T-Reverse still has no effect! Companion simulates this case wrong. On the X-lite radio you have to go to the inputs tab and invert the throttle source S1 to -100 of weight to get the timers working. On Companion you have no need to do that.

Your answer or solution is just a workaround. Why did you close my bug report? Other people are reporting and complaining the same issue. Just see the link of the frskyforum.

On my Taranis X10s Express the sliders, the timers, T-Reverse work all properly. You don‘t have to set in the Setup-tab T-Source to the Thr-Ch. That would not be logical in my opinion. Here you put in the slider. So you also don‘t have the problem with this Throttle-Warning bug. Then it could not be, that you have to deactivate the Thr-Warning with your Thr-Stick if your Thr-Input on input-tab is a Slider, on Mixers Throttle is CH3 I:Thr and on the Setup-tab T-Source is Ch03?

Thx for your help

kilrah, 2021-05-22:
Throttle warning and Throttle reverse are hardcoded to the throttle stick, it's not a bug, it's just that these aren't currently implemented in a way that allows changing the control. Could be improved in the future.

Can you post the otx that doesn't work the same in companion and radio?

frsky-forum.de requires registration to see anything for no good reason so can't see things there.

AlphaCharlyX, 2021-05-22:
Hello thank you for your help. 

Today i have opened a new issue here, because i have thought this one is closed. Sorry. Here is the new one https://github.com/opentx/opentx/issues/8524

my otx file: The difference between Model 01 and 02 is only that model 02 has Thr-Input S1 -100. 
[document2.zip](https://github.com/opentx/opentx/files/6526425/document2.zip)

AlphaCharlyX, 2021-05-24:
"_Throttle warning and Throttle reverse are hardcoded to the throttle stick, it's not a bug, it's just that these aren't currently implemented in a way that allows changing the control. Could be improved in the future._"

This could lead to a safety-problem. If you do the "workaround" for getting the timers working properly (you set T-Source to CH3-Thr and the input for Thr to S1 with weight -100 in the input tab. If the Thr-Stick is in idle-position you don't get the Thr-Warning at startup even S1 is not idle, which is defined as Thr at the input tab!

Using a slider, pot or a 3-position-switch for Thr is a usual thing for sail-plane-pilots.

3djc, 2021-05-24:
You can set a pot or slider warning instead

FAG1975, 2021-06-05:
I like to second this issue / bug!

A pot / slider normally works CW and not CCW like with the X-Lite pro and pot / slider S1. It should be just as it is with the Companion, from left to right going from -100 to +100 and not vice versa. I would be happy to get this fixed ... thank you!

bigtwin, 2021-06-05:
from @kilrah:
   Can you post the otx that doesn't work the same in companion and radio?

Its about version 2.3.11 as provided by @AlphaCharlyX  in his initial update.

In spite of any discussion whether this can be worked around and how, the behavior is supposed to be the same in _Companion_ and the transmitter itself. And this is obviously not given. 
Therefor I would also treat as a (little) bug. Either in _Companion_ or OTX on a X-lite.

3djc, 2021-06-05:
And companion is NOT targeted to match physical behaviour of radio (position, direction,…) only logical 

Also pot and sliders do not behave in the same way at all. Pots are cw, slider follow switch like rules

FAG1975, 2021-06-05:
> Also pot and sliders do not behave in the same way at all. Pots are cw, slider follow switch like rules
At the X-Lite Pro I would consider S1 a pot as it rotates like the pot S1 within Companion. Both pots (S1 and S2) should behave the same way: At the 0700 position -100 and +100 at the 0500 position (compared to a clock). Accordingly the bars at the display should run up with turning both pots CW (clock wise). If pots are programmed as throttle, the pot warning should work correspondingly and the throttle-timer should run when the motor runs. All of this can't be cured by mixer settings or the like (afaik).

janciesko, 2021-12-15:
Might be related - Mapping throttle to S1, and setting timer activation to THR still uses the left stick to activate the timer. Seems like a bug to me.

FAG1975, 2021-12-16:
> Might be related - Mapping throttle to S1, and setting timer activation to THR still uses the left stick to activate the timer. 

Hi, 
setting the throttle channel (e.g. Ch 1) as throttle timer source (instead of S1) will make the timer to work as expected.

janciesko, 2021-12-18:
> > Might be related - Mapping throttle to S1, and setting timer activation to THR still uses the left stick to activate the timer.
> 
> Hi, setting the throttle channel (e.g. Ch 1) as throttle timer source (instead of S1) will make the timer to work as expected.

Worked.