opentx/opentx #8523: Building OpenTX 2.4 under Windows 10 fails (GNU Arm Toolchain 6 2017-q2). Same GNU ARM OK under Ubuntu

rotorman, 2021-05-21:
I am trying to compile OpenTX 2.4 under Windows, but am failing miserably. I have previously experience / success with OpenTX 2.3 under Ubuntu as well as under Windows (see [here](https://www.rcgroups.com/forums/showthread.php?3824801-Blog-1-Step-by-step-guide-to-building-%28custom%29-OpenTX-firmware-under-Ubuntu-Linux) and [here](https://www.rcgroups.com/forums/showthread.php?3840589-Blog-3-Step-by-step-guide-to-building-%28custom%29-OpenTX-firmware-under-Windows-10-x64)).

I understood, for OpenTX 2.4 I need GNU Arm Toolchain in version 6 2017-q2, so have been using it for my tests.

Works flawlessly under Ubuntu 20.04, but has at least two issues, when compiling under Windows.

First issue: dependency for truetype_fonts on Horus-class radios is missing (see [EdgeTX PR #14](https://github.com/EdgeTX/edgetx/commit/590ec5a9ed79bf1ed8fc8cb99601e37c7da0b0ae), applies to OpenTX v2.4 as well).

Second issue is unfortunately not yet solved. Under Windows, the build fails for LUA lundump.c with somewhat cryptic error message (here the build output tail, when make is ran with _--trace_ option):
```[ 58%] Building C object radio/src/CMakeFiles/firmware.dir/thirdparty/Lua/src/lundump.c.obj
cd /d C:\opentx\build-24\radio\src && arm-none-eabi-gcc -DAFHDS3 -DAUDIO -DAUTOSOURCE -DAUTOSWITCH -DAUX_SERIAL -DBATTGRAPH -DCOLORLCD -DCORRECT_NEGATIVE_SHIFTS -DCPUARM -DCROSSFIRE -DCURVES -DDBLKEYS -DDEFAULT_MODE=2 -DDEFAULT_TEMPLATE_SETUP=21 -DDEFAULT_THEME_DARKBLUE -DDISK_CACHE -DDSM2 -DEEPROM_VARIANT=0 -DFLAVOUR=\"tx16s\" -DFLIGHT_MODES -DGHOST -DGPS -DGPS_USART_BAUDRATE=9600 -DGUI -DGVARS -DHAPTIC -DHARDWARE_EXTERNAL_MODULE -DHARDWARE_EXTERNAL_MODULE_SIZE_STD -DHARDWARE_INTERNAL_MODULE -DHARDWARE_KEYS -DHARDWARE_TOUCH -DHARDWARE_TRAINER_JACK -DINTERNAL_GPS -DINTERNAL_MODULE_MULTI -DLIBOPENUI -DLUA -DLUA_COMPILER -DLUA_INPUTS -DLUA_MODEL_SCRIPTS -DMANUFACTURER_RADIOMASTER -DMODULE_PROTOCOL_D8 -DMODULE_PROTOCOL_FCC -DMODULE_PROTOCOL_LBT -DMULTIMODULE -DNO_UNEXPECTED_SHUTDOWN -DOVERRIDE_CHANNEL_FUNCTION -DPCBFRSKY -DPCBHORUS -DPCBREV=TX16S -DPCBREV_TX16S -DPCBX10 -DPOPUP_LEVEL=2 -DPPM -DPPM_CENTER_ADJUSTABLE -DPPM_LIMITS_SYMETRICAL -DPPM_UNIT_US -DPWR_BUTTON_PRESS -DPXX -DPXX1 -DRADIO_FAMILY_T16 -DRADIO_TX16S -DREQUIRED_SDCARD_VERSION=\"2.4V0036\" -DRTCLOCK -DRTC_BACKUP_RAM -DRTOS_COOS -DSBUS -DSBUS_TRAINER -DSDCARD -DSDCARD_RAW -DSDRAM -DSOFTWARE_VOLUME -DSPLASH -DSTM32 -DSTM32F4 -DSTM32F429_439xx -DSTORAGE_CONVERSIONS=218 -DTHRTRACE -DTIMERS=3 -DTRACE_SIMPGMSPACE -DTRANSLATIONS=\"EN\" -DTRANSLATIONS_EN -DUSB_CHARGER -DUSB_SERIAL -DUSE_BIN_ALLOCATOR -DVARIO -DVOICE -DWATCHDOG -DXJT -D_GLIBCXX_USE_C99=1 -IC:\opentx\otx24\radio\src\thirdparty\STM32F4xx_DSP_StdPeriph_Lib_V1.8.0\Libraries\STM32F4xx_StdPeriph_Driver\inc -IC:\opentx\otx24\radio\src\thirdparty\STM32F4xx_DSP_StdPeriph_Lib_V1.8.0\Libraries\CMSIS\Device\ST\STM32F4xx\Include -IC:\opentx\otx24\radio\src\thirdparty\STM32F4xx_DSP_StdPeriph_Lib_V1.8.0\Libraries\CMSIS\Include -IC:\opentx\otx24\radio\src\targets\common\arm\stm32 -IC:\opentx\otx24\radio\src\thirdparty\STM32_USB-Host-Device_Lib_V2.2.0\Libraries\STM32_USB_OTG_Driver\inc -IC:\opentx\otx24\radio\src\thirdparty\STM32_USB-Host-Device_Lib_V2.2.0\Libraries\STM32_USB_Device_Library\Core\inc -IC:\opentx\otx24\radio\src\thirdparty\STM32_USB-Host-Device_Lib_V2.2.0\Libraries\STM32_USB_Device_Library\Class\msc\inc -IC:\opentx\otx24\radio\src\thirdparty\STM32_USB-Host-Device_Lib_V2.2.0\Libraries\STM32_USB_Device_Library\Class\hid\inc -IC:\opentx\otx24\radio\src\thirdparty\STM32_USB-Host-Device_Lib_V2.2.0\Libraries\STM32_USB_Device_Library\Class\cdc\inc -IC:\opentx\otx24\radio\src\thirdparty\CoOS -IC:\opentx\otx24\radio\src\thirdparty\CoOS\kernel -IC:\opentx\otx24\radio\src\thirdparty\CoOS\portable -IC:\opentx\otx24\radio\src\targets\horus -IC:\opentx\otx24\radio\src\thirdparty -IC:\opentx\otx24\radio\src -IC:\opentx\build-24\radio\src -IC:\opentx\build-24\radio\src\bitmaps\480x272 -IC:\opentx\otx24\radio\src\gui\colorlcd -IC:\opentx\otx24\radio\src\gui\colorlcd\layouts -IC:\opentx\build-24\radio\src\fonts -IC:\opentx\otx24\radio\src\thirdparty\libopenui\src -IC:\opentx\otx24\radio\src\gui\libopenui -IC:\opentx\otx24\radio\src\gui -IC:\opentx\otx24\radio\src\thirdparty\Lua\src -IC:\opentx\otx24\radio\src\thirdparty\FatFs -IC:\opentx\otx24\radio\src\thirdparty\FatFs\option -IC:\opentx\build-24\radio\src\targets\common\arm\stm32\bootloader -mcpu=cortex-m4 -mthumb -fomit-frame-pointer -fverbose-asm -Wa,-ahlms=firmware.lst -Os -gdwarf-2 -DHSE_VALUE=12000000 -fno-exceptions -fdata-sections -ffunction-sections  -Wall -Wno-address-of-packed-member -Wno-strict-aliasing -Wformat -Wreturn-type -Wunused -Wuninitialized -Wunknown-pragmas -Wno-switch -Wtype-limits -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wimplicit -g   -g -o CMakeFiles\firmware.dir\thirdparty\Lua\src\lundump.c.obj -c C:\opentx\otx24\radio\src\thirdparty\Lua\src\lundump.c
C:\Users\rotorman\AppData\Local\Temp\ccpwm7ds.s: Assembler messages:
C:\Users\rotorman\AppData\Local\Temp\ccpwm7ds.s: Warning: end of file in comment; newline inserted
C:\Users\rotorman\AppData\Local\Temp\ccpwm7ds.s: Error: open CFI at the end of file; missing .cfi_endproc directive
mingw32-make[3]: *** [radio\src\CMakeFiles\firmware.dir\build.make:1511: radio/src/CMakeFiles/firmware.dir/thirdparty/Lua/src/lundump.c.obj] Error 1
mingw32-make[3]: Leaving directory 'C:/opentx/build-24'
mingw32-make[2]: *** [CMakeFiles\Makefile2:408: radio/src/CMakeFiles/firmware.dir/all] Error 2
mingw32-make[2]: Leaving directory 'C:/opentx/build-24'
mingw32-make[1]: *** [CMakeFiles\Makefile2:415: radio/src/CMakeFiles/firmware.dir/rule] Error 2
mingw32-make[1]: Leaving directory 'C:/opentx/build-24'
mingw32-make: *** [Makefile:175: firmware] Error 2
mingw32-make: Leaving directory 'C:/opentx/build-24'```

Would you have any pointers where to look for an error?

bsongis, 2021-05-21:
out of memory?

bsongis, 2021-05-21:
mingw makefiles and msvc2019?

rotorman, 2021-05-21:
Thank you for a swift answer. I have 8GB RAM, during build I see 4GB free, but maybe 32-bit process, so can only access 2GB and hits this limit. Would you have a tip how to check this during build?

rotorman, 2021-05-21:
> mingw makefiles and msvc2019?

Sorry I did not get exactly how can I answer - did you mean that this is a problem, or did you ask me to provide some files?

My system is setup as here: [https://www.rcgroups.com/forums/showthread.php?3840589-Blog-3-Step-by-step-guide-to-building-%28custom%29-OpenTX-firmware-under-Windows-10-x64](https://www.rcgroups.com/forums/showthread.php?3840589-Blog-3-Step-by-step-guide-to-building-%28custom%29-OpenTX-firmware-under-Windows-10-x64) with only difference that I am using for OpenTX 2.4 GNU Arm Toolchain 6 2017 q2 instead of 4.7 2013 q3 as listed on that page for OpenTX 2.3

rotorman, 2021-05-21:
I changed with MinGW 8.10 64-bit. Unfortunately build fails at the same point as before with 32-bit.
Visual Studio Enterprise 2019 is installed.
@bsongis Would you have further tips?

rotorman, 2021-05-21:
I installed in Qt, the MSVC2019 64-bit (instead of 32-bit) and re-tried with this, unfortunately results in the same error message.

rotorman, 2021-05-21:
Building with newer GNU ARM Toolchains than 6 2017q2 succeeds under Windows, but they do not run correctly on radio (TX16S tested), when LUA is included (CMake -DLUA=YES, Hardfaults due to alignment error near Lua lstring.c _createstrobj()_).