opentx/opentx #8524: Taranis X-lite Pro using Sliders as Throttle you get false Throttle-Warning, Timer don't start, T-Reverse has no effect

AlphaCharlyX, 2021-05-22:
Hello together,

there is a really a problem with starting the Timers (Ths, TH% and THt) probably, if you are using the S1- or S2-Slider on your X-lite Pro. Throttle-On should start the timer and not Throttle-Off-Position of S1 or S2). In addition to that T-Reverse doesn't work with any of the S1- or S2-Sliders.

On Companion and on my X10s Express everthing works right as it should. There it is not a problem to define T-Source to the sliders S1 or S2 at the Setup-tab. That is the way it is logical and should work.

On the contrary to that on the X-lite Pro you have to do a "Workaround" as shown with the pictures below): You have to set T-Source at the setup-tab to your Thr-Channel (CH3). In additon to that on the X-lite pro radio you have to go to the Inputs-tab and invert the weight at Throttle-Source S1 to -100 to get the timers working properly. On Companion itself you have no need to do that step of inverting.

The problem with this "Workaround" is, that you get a Throttle-Warning at the Start-up of the X-lite pro if - and that is really strange and for me not comprehensible - the "Throttle-Stick" and not the for Throttle defined Slider S1 or S2 is not in the idle-position! 

Other people are reporting and complaining the same issue. Just see in ther german frskyforum.

On my Taranis X10s Express the sliders, the timers, T-Reverse work all properly. You don‘t have to set in the Setup-tab T-Source to the Thr-CH. Here you just put in the chosen slider for Throttle. You also have no problems with Throttle Warnings if the chosen Slider is in the idle-position.

It would be nice if you could have a closer look to this annoying bug.

    OpenTX version (e.g. 2.3.11 or 4328055)
    Hardware Frsky Taranis X-lite Pro
    OpenTX Companion 2.3.11 on Win10

These are my settings:

[image]
[image]
[image]

3djc, 2021-05-22:
xlite does have pots, you cannot compare that with x10 sliders (they do not behave in similar ways)

AlphaCharlyX, 2021-05-22:
my otx file: The difference between Model 01 and 02 is only that model 02 has Thr-Input S1 -100.
[document2.zip](https://github.com/opentx/opentx/files/6526428/document2.zip)

kilrah, 2021-05-22:
Duplicate #8520

closed 2021-05-22 by kilrah